OPCFoundation/UA-Java-Legacy #21: Private key storage on Android private storage fails (Mantis 2337)

dk2129, 2016-01-22:
The PrivKey can be stored in PEM format on the sdcard (ExternalStorage), but not in the internal storage. Reading it from there fails for some reason.

It would be safer to store the private key in the internal memory which only the application can access itself.

jouniaro, 2017-07-21:
Strong the key seems to work on the latest Android versions, but now the PrivateKey is loaded in a format that cannot be used to sign asymmetric messages leading to errors in ScCryptoProvider.signAsymm. Needs to be checked why the PrivateKey is not of a usable instance and how to make it compatible again. This is possibly a change in the recent Android versions.

jouniaro, 2017-10-17:
This is not a real problem any more, so will close.

closed 2017-10-17 by jouniaro