OPCFoundation/UA-Java-Legacy #25: Several endpoints for SecurityMode.None returned, if several certificates defined (Mantis 2690)

dk2129, 2016-01-22:
The Java stack is capable of handling several certificates, but it seems to return several endpoints for the same security settings for each certificate, especially for SecurityMode.None, which has no limits for the certificate key size (other policies define the valid keysize range).

Also, if no user token policies, except for the Anonymous would be defined, the server certificate is defined for the endpoints - although that was not supposed to be the case.

In practice, this can lead to user password decryption failure, if the client cannot use a big certificate that may be returned fro the endpoint, to be used for password encryption.

jouniaro, 2016-09-29:
Since several certificates are not used much in practice (at least yet), we can keep this as low priority for now

bjakke, 2018-11-13:
This is a side-effect of how the endpoints are defined, currently it is loop over all certs and make endpoints (based on the configuration) that can be done with that certificate (if configured; 'None' can be made with any cert size). 

There should be a separate logic for defining the None endpoint that does not define a certificate at all (that would also validate that it only uses usertokenpolicies not requiring encryption). However it should be noted that for interop with UA 1.01 clients this should preferably be never used if there is an option to use a one with the certificate, i.e. only if the server is incapable of making a cert at all, then this could be an option.

It should be checked on how the situation should be handled if there are multiple endpoints for None that would require the cert (e.g. for password encryption), as to my understanding it is allowed to have multiples of them. In order to be interop with as many clients as possible, there might be valid use-cases to have more than one (if the usertokenpolicies require certs). For example dealing with a client only supporting 1024 cert sizes and another only supporting 4096.

This can be workarounded by trying to decrypt the password with each of the certs the server is using in turn.