OPCFoundation/UA-Java-Legacy #43: Recommended development environment

jwende, 2016-06-08:
Hi,
I wonder which versions of Eclipse, Maven, etc are recommended for development ?
Thanks
Joerg

bjakke, 2016-06-08:
Good point. 

I'm using Eclipse Mars (release 2) 4.5.2 ("Eclipse IDE for Java Developers" edition). It's m2e maven integration has maven version 3.3.3 embedded. So 3.3.3 and later should work at least. I'm not sure about what is the minimum versions of maven needed for the plugins. Needs to be checked and added to README when known (also maybe maven-enforcer-plugin to pom).

Now that the project is a maven project (#32), it should be usable via any IDE that supports maven projects, so e.g. NetBeans and IDEA should also work (I have not tested this yet). The eclipse project files are in the repo because they where there before the conversion, but I guess removing them could be also an option. Also at some point checkstyle or similar style checking could be added.

jwende, 2016-06-08:
Thank you !
Just went over the the readme.md to get more details - are you really sure that I need a JRE 1.6 to compile the project ??
J.

bjakke, 2016-06-08:
You can disable the maven-toolchains-plugin via disabling the 'jdk6' profile (from IDE or add '-P !jdk6' to mvn cmd command). You should then be able to use later ones (well only upto future 9, see http://openjdk.java.net/jeps/182 ). 

Note that if you do not compile it with JDK6, then the standard library which the classes are "linked" is the one which you are using to run maven, i.e. if you use e.g. JDK7, then the resulting jars wont necessarily work on Java 6. 

See e.g. http://stackoverflow.com/questions/34755121/issue-about-java-8-backward-compatibility-new-methods-in-jdk

bjakke, 2016-06-08:
The answer is this stackoverflow post explains the issue a bit better:
http://stackoverflow.com/questions/35913775/maven-java-version-configuration-ignored-by-eclipse-idea/

As a sidenote, this project also uses the maven-animal-sniffer plugin mentioned in that stackoverflow answer. In theory that should be enough when using JDKs that can compile to 6, but because the plugin relies on checks, it can and has issues (see https://github.com/mojohaus/animal-sniffer/issues ). In my opinion it is best to use the toolchains-plugin to to be 100% sure that the classes are compiled as Java 6 compatible. Also I would assume the need for this project to be Java 6 compatible for the for the unforeseeable future so after java 9 (in 10+, see the jep 182 link in earlier) we need to use the toolchains anyway.  

The one thing that could be considered is to disable the jdk6 profile by default (currently active by default) to allow easier use for use-cases not requiring java 6 compatible.

jwende, 2016-06-09:
Thank you so far - will give it a try.
J.

commit 97458398 (2016-12-09): Fixed #43, updated sections about 'jdk6' profile
closed 2016-12-09 by bjakke