OPCFoundation/UA-Java-Legacy #46: Maven build issues

mregen, 2016-06-24:
I ran into a few build issues with maven cmd line build on windows:

it only succeeded after disabling test and javadoc

mvn package -DskipTests=true -Dmaven.javadoc.skip=true

Please repair.

bjakke, 2016-06-24:
Could you please post the errors? It is hard to do anything without more info. 
Also for reproducing I need the versions of maven and Java you are using, both the JDK 6 and the one you are using to run maven (since your cmd does not show disabling the jdk6 profile which is active by default and assuming you are using relatively new version of maven which cannot be run on 6).

With the current state of javadocs they will give some warnings but should still work. Which reminds me that when I made the conversion to a maven project if the build was done using Java 8 it had stricter error checking and needed the `-Xdoclint:none` parameter. However on Java 6 that flag was unknown and resulted build failing. With the toolchains plugin this was no longer a problem (the flag is still commented out in the POM), but now that there is an option to disable the toolchain/java6 via profiles it should maybe be taken into consideration.

mregen, 2016-06-26:
@bjakke Thanks for your quick response. The problem I'm facing is that JDK6 for windows isn't available for download anymore, only to Oracle customers. So I must use my latest Java version to compile the whole thing (set in toolbox.xml). I'm using windows jdk1.8.0_92 + Maven 3.3.9.
Please let me know if you need more info.

bjakke, 2016-06-27:
Hmm at least the archive downloads http://www.oracle.com/technetwork/java/javase/downloads/java-archive-downloads-javase6-419409.html  work for me (though you do need to register a free oracle account). And for anyone installing: Note the warnings on that page. You'll want to uncheck the Public JRE option when installing the JDK. This way only the Private JRE for the JDK is installed (to my knowledge it should not modify PATHs or install the browser plugin). So you can use it only for development/testing.

Still that probably is not the easiest way to start working with the stack.. and this is a second issue (see #43) that would point using latest instead of 6 for development... so I suppose we have to move the toolchain/jdk6 requirement to a profile that is by default off and document it well. I suppose that the animal-sniffer-maven-plugin together will source and target 1.6 options for the compiler plugin would ensure enough checking to keep Java 6 compliant during development. Maybe the toolchain plugin for development was after all a bit overkill. I would still use it for building the final artifact in case it needs to work on Java 6 to be 100% sure, see the #43 later comments for why. But I suppose it should be made optional as in many cases I would assume separate CI/builder doing the final build so it shouldn't be a problem to use separate profiles, i.e. developers would not have change their settings/cmdlineargs that which profiles are used and there would still be option to do so if needed. It just needs to be very well documented that the default artifact of the build would not be 100% certain Java 6 compatible without the profile/toolchain on (they are more or less corner cases, but might still be relevant). 

Javadocs needs to be fixed so that they can be created in 8, or a profile for the error-disable flag in short term.

Still need to check the tests.. if possible, can you what problems you had with those?

mregen, 2016-07-01:
@bjakke I'm on vacation until mid July but will check when I'm back. Thanks for your kind advise in the meantime.

bjakke, 2016-12-09:
Fixed in referenced commit. I was not able to reproduce any issues with the tests and #47 fixes javadoc problem, therefore closing this issue.

#46, #48, refactored pom to have a jdk6 profile that is by default not
on (see #47)

closed 2016-12-09 by bjakke