OPCFoundation/UA-Java-Legacy #49: Remove NodeClass.Unspecified

jouniaro, 2016-06-30:
Since Unspecified is not defined in the specification, we should also remove this identifier from the stack. 'null' value should be used instead to define an undefined NodeClass - for consistency with other enumerations in the stack.

However, it can be a breaking change, if applications have been using it to detect an undefined, so further consideration should be given.

bjakke, 2016-12-09:
Fixed in referenced commit, was not used by the stack.

bjakke, 2017-01-13:
Actually the Opc.Ua.NodeSet2.xml for the standard model even as of 1.03 version does define the NodeClass.Unspecified at value 0, therefore reopening this issue as we might need to revert this change. 

It probably is an error in the model as specification does not define this value, but probably reverting this for now is the easiest solution until a model version comes that does not define the Unspecified value.