OPCFoundation/UA-Java-Legacy #50: Standard Engineering Units

jouniaro, 2016-06-30:
Import the standard engineering units as defined in Spec. 1.03 Part 8 into a helper class, such as EngineeringUnits. The class could be code generated using the CSV file available from

https://opcfoundation.org/UA/EngineeringUnits/UNECE/UNECE_to_OPCUA.csv

This will define a list of EUInformation objects - that should be in a map that enables searching for the object using the UnitId - and possibly the DisplayName and Symbol as well.

jouniaro, 2016-07-15:
The encoding of the CSV is not quite clear. There are characters, such as Ohm that are not displayed by editors (e.g. Notepad++) in ANSI mode. Or is it just a font issue?

Or, actually, it looks like these characters are written as '?' in the cvs file.

jouniaro, 2016-07-15:
Created a Mantis task for that:
https://opcfoundation-onlineapplications.org/mantis/view.php?id=3471

bjakke, 2016-12-09:
Fixed in referenced commit. In the end I needed to create new export of the UNECE excels as utf-16, it is now part of the codegen resources and used to create the StandardEngineeringUnits class. NOTE! if there is a duplicate description of a standard unit, then the dupcates are ignored as the UNECE seems to not have any programming-language-friendly name, therefore the current implementation parses the name from the description field (it is easier to use than having the constant to be e.g. COMMON_CODE_??), convers all characters to upper case, removes/replaces some characters that cannot be used in java constants and replaces spaces with underscores.

#50: Added support for StandardEngineeringUnits, see it's constants.
Note that the constant naming is a best effort of parsing the
description

closed 2016-12-09 by bjakke