OPCFoundation/UA-Java-Legacy #51: Need advise how to integrate OPC Publihser sample

mregen, 2016-07-01:
Hi,
I'm working on a OPC Publisher sample as contribution: https://github.com/mregen/UA-Java/commit/91f126dbdc1df8fa3450109849fea07434784abb
Now given the Maven build process my difficulty is a dependency on some external jar libs and I figured the place under examples may not be the best fit. Do you have advise where you would add such a sample and how to incorporate in the maven build process? My current thinking is to have a separate Maven build due to the dependencies (https://github.com/Azure/azure-iot-sdks/tree/master/javawrapper requires a manual build step to build the .jar for Azure IoT SDK integration).

Thanks,
Martin

bjakke, 2016-07-01:
Are all needed jars of these: http://mvnrepository.com/search?q=com.microsoft.azure.iothub  or if not would it be possible to put the rest to maven central too? That removes the need to build the jar for users of the sample. 

I would put this sample under a different folder and have different pom as the current examples because of different dependencies. Assuming there are multiple sample contributions in the future, it would probably be best to move the current example to a subfolder (not sure about name for it yet). 

So folder structure would be like

- root/examples
 - 'createagoodname' (current examples)
   - (pom.xml for those examples)
 - azure-integration
   - (pom.xml for azure specific needs)

bjakke, 2016-07-06:
This can be done after #54 is done.

mregen, 2016-07-12:
@bjakke thanks for your advise, I can refactor the sample to have the dependency on the Azure IoT Device SDK in the Maven central.

closed 2016-07-12 by mregen