OPCFoundation/UA-Java-Legacy #58: Please rename Azure-publisher sample to "Publisher"

barnstee, 2016-07-21:
This will make it consistnent with the other Publisher samples and the sample is not Azure-specific.

bjakke, 2016-07-21:
Are you absolutely sure you want the folder to be "Publisher" and not "publisher". I would not like to have one folder with different casing (all others are lowercase) based on the sole reason that it would be exactly same as in other stacks as different programming languages have different conventions. For example in .NET the namespaces use Pascal casing (https://msdn.microsoft.com/en-us/library/893ke618%28v=vs.71%29.aspx) whereas in Java packages they are all in lowercase. While this convention is not enforced/mandatory for other folder structures in a project I would prefer to keep uniform style within a project and currently all other folders in this project are all in lowercase and files are kept as close as possible to maven conventions (https://maven.apache.org/guides/introduction/introduction-to-the-standard-directory-layout.html).

barnstee, 2016-07-21:
"publisher" is fine.

commit 903ffe31 (2016-07-22): Fixed #58, renamed 'azure-publisher' to 'publisher'
closed 2016-07-22 by bjakke