OPCFoundation/UA-Java-Legacy #63: License texts are different in LICENSE.txt and on official page

AriSuutariST, 2016-09-12:
License in http://opcfoundation.github.io/UA-Java/ states that RCL license applies to "OPC Foundation Corporate Members". 

License in https://github.com/OPCFoundation/UA-Java/blob/master/LICENSE.txt states that RCL applies to "OPC Foundation members in good-standing".

These sound different to me. To my understanding, the wording in LICENSE.txt allows RCL licensing to OPC foundation end users also.

Which one is correct ?

jouniaro, 2016-09-28:
To "OPC Foundation Corporate Members" is correct. LICENSE.txt should changed accordingly.

AriSuutariST, 2016-09-29:
Ok, good to know :-(

jouniaro, 2016-09-29:
Fine, but let's keep it open until the text has been changed.

bjakke, 2016-12-09:
Fixed in referenced commit.

closed 2016-12-09 by bjakke