OPCFoundation/UA-Java-Legacy #64: UA Java Stack cannot reconnect.

nakasuke, 2016-09-14:
I use OPC UA Java stack ,but I have a unknown error to communication between Client and Server.
The error is the contents of the following.

SDK Version:1.03.340
Platform:Windows 7

A OPC-UA Client connect OPC-Classic Server via OPC COM Wrapper.
If you turn off the network that is connected between the OPC-UA
client and server to force, OPC-UA client is unable to reconnect.

error log message is
”[TcpConnection/Read] WARN org.opcfoundation.ua.transport.tcp.io.TcpConnection -
/10.1**.1**.151:6**47 OpenSecureChannel, server sent two secureChannelIds 769 and 776 using 769”

Best regards.

jouniaro, 2016-09-29:
This is just a warning message and should not affect the communication. But in practice, the reconnection must be made on the application level, making sure that the secure channel and session are recreated as necessary.

nakasuke, 2016-10-01:
Thank for your answer. I was discouraged because a while comments on this article did not stick.

Since the re-connection was overconfidence that the communication stack us be carried out in the back, I do not write code to reconnect at the application level. For studying the shortage, there is also that the code to be re-connected did not know. I think let's write the code the sample program that OPC-F is provided to the reference.

jp2rtq, 2016-10-01:
Thank you for your information.

I'm having the same problem.
I created a simple application that create SessionChannel and wait forever.

When I disconnect and reconnect the network, UA Java stack automatically executes SecureChannelTcp.reconnectRunnable in subthread.
It tries to reconnect and fail repeatedly, and  it increases secureChannelIds. The Log is below.

```
// disconnect and reconnect.
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Closed (unexpected)
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Closed
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - 65: Error recovery = true
[Blocking-Work-Executor-1] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Connecting
[Blocking-Work-Executor-1] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Connected
[TcpConnection/Read] WARN org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 OpenSecureChannel, server sent two secureChannelIds 65 and 66 using 65
[Blocking-Work-Executor-1] INFO org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - 65: Error recovery = false

// repeat
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Closed (graceful)
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Closed
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - 65: Error recovery = true
[Blocking-Work-Executor-3] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Connecting
[Blocking-Work-Executor-3] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Connected
[TcpConnection/Read] WARN org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 OpenSecureChannel, server sent two secureChannelIds 65 and 67 using 65
[Blocking-Work-Executor-3] INFO org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - 65: Error recovery = false
[TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /XXX.XXX.XXX.XXX:48400 Closed (graceful)
// repeat
```

I want to suppress java stack's auto re-connection.
Is there any sample code for re-connection in java stack?

jouniaro, 2016-10-06:
The stack's reconnection works automatically in the background. But it may be necessary to compliment it on the application level. Not quite sure about the exact behaviour here; why the secure channel keeps recreated.