OPCFoundation/UA-Java-Legacy #68: Android Tutorial or sample project

haru-a8n, 2016-10-14:
Document Android Tutorial or provide sample project targeting Android platform. I have spent almost a week of late night work trying to figure out how to make the SDK (client) work on Android. Note that I am no expert in Java nor Android.

I am willing to start Android tutorial if someone can provide me some directions. (* I am already spending time on it... so no big deal. Just provide hints and I will try to get it up and running*)

ChristophRathfelder, 2017-02-02:
Have you been able to run the UA stack on android? I also want to use it within an android app, but can not estimate the effort required to setup a project with all dependencies and maybe adding some code fixes.

jouniaro, 2017-02-02:
The stack works properly with Android as is demonstrated by the Prosys OPC UA Client for Android. You don't need to modify anything in principle. It will also adapt to use the Spongy Castle crypto library automatically, as mentioned in the [Readme](https://github.com/OPCFoundation/UA-Java/blob/master/README.md)

closed 2017-10-17 by jouniaro