OPCFoundation/UA-Java-Legacy #70: unable to connect to most opc ua servers

makraf1, 2016-11-18:
I used SampleClient to connect to many UA Servers but only CommServer give the possibilty to accept or not the certificate for the other servers I get Exceptions

jouniaro, 2016-11-22:
Would help, if you would be more specific about the issue. Which server, for example and which exception?

makraf1, 2016-11-23:
I passed this argument (opc.tcp://hostname:port /Advosol/**uaPLUSsim**) in SampleClient I get this exception

4455 INFO  [TcpConnection/Read] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - IO-DELL-PC4/192.168.0.199:62849 Closed (graceful)
4456 INFO  [TcpConnection/Read] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - IO-DELL-PC4/192.168.0.199:62849 Closed
5466 DEBUG [main] org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp     - requests.clear()
Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_ConnectionClosed (code=0x80AE0000, description="Connection closed (graceful)")
	at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run(TcpConnection.java:967)
Caused by: java.io.EOFException
	at org.opcfoundation.ua.utils.bytebuffer.InputStreamReadable._get(InputStreamReadable.java:60)
	at org.opcfoundation.ua.utils.bytebuffer.InputStreamReadable.getInt(InputStreamReadable.java:141)
	at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run(TcpConnection.java:720)

jouniaro, 2016-11-23:
Sorry, I don't have any experience of Advosol servers. The error sounds like there is no good response from that address (hostname:port). Better to verify that the server address is correct and that you can connect to the server address with some free OPC UA client first. Prosys OPC UA Client uses the same stack, for example. UaExpert is the one that most people use to test connections. If the server is compliant, it should work with both.

makraf1, 2016-11-23:
I tried opc.tcp://IO-DELL-PC4.iodomain.com:53530/OPCUA/SimulationServer
I get 
959  WARN  [TcpConnection/Read] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - IO-DELL-PC4.iodomain.com/192.168.0.199:53530 Error
org.opcfoundation.ua.common.ServiceResultException: Bad_SecurityChecksFailed (code=0x80130000, description="Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security.")")
	at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run(TcpConnection.java:763)
962  INFO  [TcpConnection/Read] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - IO-DELL-PC4.iodomain.com/192.168.0.199:53530 Closed
2013 DEBUG [main] org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp     - requests.clear()
Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_SecurityChecksFailed (code=0x80130000, description="Bad_SecurityChecksFailed (code=0x80130000, description="An error occurred verifying security.")")
	at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run(TcpConnection.java:763)

jouniaro, 2016-11-23:
That is expected. You need to go to the server configuration and trust the certificate of the client. The SimulationServer has a Certificates tab in the UI for that.

makraf1, 2016-11-23:
I trusted the certificate and re run he throw the same exception and add another rejected certification with same name

jouniaro, 2016-11-23:
Ah, true. The sample client is not capable of storing the certificate in a file and therefore creates a new certificate every time you run it.

This is as defined at the moment, although not very nice in practice. The sample client is extremely limited.

jouniaro, 2016-11-23:
I suggest you change the security mode to None in the code. Then you should be able to connect.

jouniaro, 2016-11-23:
See also: #8

closed 2016-11-24 by jouniaro