OPCFoundation/UA-Java-Legacy #71: org.opcfoundation.ua.application.Application.Server

gabun88, 2016-11-30:
The server loads a default discovery service (the onGetEndpoints method more precisely) that cannot be replaced. Constructor is adding a service to the handler and the handler has no remove method.
Is there a reason for this behaviour? 

There is a second problem with this class, with I ended up using to assigned a new discovery service, but I find that this is not the correct way of handling this. 

The constructor is calling a overridable method (addServiceHandler). 

     public Server(Application application) {
        this.application = application;
	    endpointBindings = new EndpointBindingCollection();
	    endpointDiscoveryService = new EndpointDiscoveryService( endpointBindings );
        //Calling a public overridable method.
	    addServiceHandler( endpointDiscoveryService );		
    }

A quote from Effective Java 2nd Edition, Item 17: Design and document for inheritance, or else prohibit it:

> There are a few more restrictions that a class must obey to allow inheritance. Constructors must not invoke overridable methods, directly or indirectly...

jouniaro, 2016-12-01:
I guess the reason is to ensure that the default implementation for onGetEndpoints is always present in the server.

You can overcome the issue by adding a service handler that adds the service methods that you wish to implement without implementing the whole interface.

The stack design does carry some legacy.

bjakke, 2017-10-17:
The work on stack started around when Java 6 was just released, therefore some of the APIs might not reflect modern design principles of today. There is a lots that could be modernized, however it is hard to do that without breaking the API too much.