OPCFoundation/UA-Java-Legacy #76: 'Package' target fails on tests

miraenator, 2016-12-20:
Hi, 

(appplies both to 'release' and 'master')

I have an issue compiling the stack. The closest I get is the 8 errors (tests). The problem is that in such case the 'mvn package' fails and does not create jars. (Trying to use the stack in a docker container, i.e. automatically git clone && mvn package.)

I've tried many combinations: with/without jdk6 target, installing bc crypto provider manually, java7/8, ...

I think this is an issue that might block newcomers from using your project. If they try to install/use the stack (simply) using

```git clone ; mvn package```

There are errors and no JAR files that can be used directly are not created. Although the compilation phase succeeds.

IMHO: You should either turn-off (ignore or print skipped message) the failing tests, or note it explicitly at the Readme.md page so that newcomers end-up with a working (compile/package) project.

This also applies to the release. I would expect the release to pass the tests. 

Thanks for fixing, 

    m.

The (automatically generated HTML result of the tests is attached).
[Test Results - All_in_UA-Java.html.zip](https://github.com/OPCFoundation/UA-Java/files/663452/Test.Results.-.All_in_UA-Java.html.zip)

bjakke, 2016-12-20:
Thanks for the results. This is a problem and needs to be fixed, however it might take time to reproduce.

Which operating system and jvm vendor (oracle, openjdk) did you use? On windows 7 64bit and oracle java 6 update 43 all tests do pass (on every and multiple runs).

bjakke, 2016-12-20:
A quick test this time on windows 10 64bit : 8u40 all pass, 8u60 testHttps_TLS_1_0 fails (others pass)

```
org.opcfoundation.ua.common.ServiceResultException: Bad_CommunicationError (code=0x80050000, description="2147811328, No appropriate protocol (protocol is disabled or cipher suites are inappropriate)")
	at org.opcfoundation.ua.transport.https.HttpsClientPendingRequest.run(HttpsClientPendingRequest.java:212)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: javax.net.ssl.SSLHandshakeException: No appropriate protocol (protocol is disabled or cipher suites are inappropriate)
	at sun.security.ssl.Handshaker.activate(Handshaker.java:503)
	at sun.security.ssl.SSLSocketImpl.kickstartHandshake(SSLSocketImpl.java:1482)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1351)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1403)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1387)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:553)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:412)
	at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:179)
	at org.apache.http.impl.conn.ManagedClientConnectionImpl.open(ManagedClientConnectionImpl.java:328)
	at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:612)
	at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:447)
	at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:884)
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:117)
	at org.opcfoundation.ua.transport.https.HttpsClientPendingRequest.run(HttpsClientPendingRequest.java:152)
	... 3 more
```
with 8u111 I got same results as 8u60. So maybe support for tls 1.0 is dropped in the newer jdks, needs to be investigated. But anyway this seems to be a problem with the platform. All the errors in your report are from HTTPS/TLS tests so are you sure your platform supports them? With quick googling I found e.g. https://bugs.launchpad.net/ubuntu/+source/openjdk-7/+bug/1314113. As the https transfer protocol is not yet common, you could ignore those tests in your build for immediate workaround.

miraenator, 2016-12-21:
Hi, thanks for pointing out, i'll try to take a look at TLS/Java,
My goal is to use the Java Stack on Raspberry Pi 

(BTW: Oracle Java6 is no more available for RPi/Raspbian -- install fails, as the JDK6/ARM is not available for download from Oracle). 

I have tested various JDKs under Docker (from Debian Jessie), I'm summarizing system info and results below. In general, there are 8 errors (TLS 1.0/1.1/1.2) related. Oracle JDK9 does ont compile.

## Tests Performed: 
- SSL Supported/Enabled protocols in Java (default installation)
- compile/package UA-Java/master: mvn package

## Tested Java SDKs:
- Oracle JDK 6 without unlimited-jce (not avail via repository)
- Oracle JDK 7 w. unlimited-jce
- Oracle JDK 8 w. unlimited-jce
- Oracle JDK 9 w. unlimited-jce
- OpenJDK 7
- OpenJDK 8

## Result summary:
### Protocol support:
```
JDK (deb jessie)    Supported/ Enabled    SSLv2Hello  SSLv3 TLSv1 TLSv1.1 TLSv1.2
Oracle 1.6.0_45      3/3                      SE       SE     SE     -        -
Oracle 1.7.0_80      5/1                      S        S      SE     S        S
Oracle 1.8.0_111     5/3                      S        S      SE     SE       SE
Oracle 1.9-ea        5/3                      S        S      SE     SE       SE
OpenJDK 1.7.0_111    5/1                      S        S      SE     S        S
OpenJDK 1.8.0_111    5/3                      S        S      SE     SE       SE
```

### Compilation of UA-JAVA (master/HEAD)
The number of errors vary (not sure for the cause, tests in-prog) from about 32/21 to 8. 
Does not compile: Oracle 1.9
Other JDKs (Oracle 6/7/8, Open 6/8) compile correctly, at least once ended with the following errors:
```
Tests in error:
  TestMaxMessageSize.test_Client_Https_TLS_1_0 ? ServiceResult Bad_Communication...
  TestMaxMessageSize.test_Client_Https_TLS_1_1 ? ServiceResult Bad_Communication...
  TestMaxMessageSize.test_Client_Https_TLS_1_2 ? ServiceResult Bad_Communication...
  TestMaxMessageSize.test_Server_Https_TLS_1_0 ? ServiceResult Bad_Communication...
  TestMaxMessageSize.test_Server_Https_TLS_1_1 ? ServiceResult Bad_Communication...
  TestMaxMessageSize.test_Server_Https_TLS_1_2 ? ServiceResult Bad_Communication...
  TestStack.testHttps_TLS_1_0 ? ServiceResult Bad_CommunicationError (code=0x800...
  TestStack.testHttps_TLS_1_1 ? ServiceResult Bad_CommunicationError (code=0x800...

Tests run: 164, Failures: 0, Errors: 8, Skipped: 1
```
## System info: Debian Jessie, docker container 'FROM debian:jessie'
```
# uname -a
Linux 275ed4598f25 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u2 (2016-10-19) x86_64 GNU/Linux

# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.6 (jessie)
Release:        8.6
Codename:       jessie

mvn --version
Apache Maven 3.0.5
```

---
## Installation notes: Oracle: Added repository:
```
### For Oracle-JDKs:
# cat /etc/apt/sources.list.d/webupd8team-java.list 
deb http://ppa.launchpad.net/webupd8team/java/ubuntu precise main
[... 125 lines cut ...]
  SSLv2Hello
  SSLv3
  TLSv1
  TLSv1.1
  TLSv1.2
Enabled Protocols: 1
  TLSv1
```
# OpenJDK 1.8
```
# java -version && java ProtocolTest
openjdk version "1.8.0_111"
OpenJDK Runtime Environment (build 1.8.0_111-8u111-b14-2~bpo8+1-b14)
OpenJDK 64-Bit Server VM (build 25.111-b14, mixed mode)
Supported Protocols: 5
  SSLv2Hello
  SSLv3
  TLSv1
  TLSv1.1
  TLSv1.2
Enabled Protocols: 3
  TLSv1
  TLSv1.1
  TLSv1.2
```

jouniaro, 2017-01-05:
There is some issue with Linux that the previous tests using HTTPS do not clean the system properly and the next text fails simply because the server is not accessible. So it is not directly related to the TLS-support, but rather to the test suite.

As @bjakke reports, it works better in Windows.

But, this needs to be investigated and fixed as well.

miraenator, 2017-02-09:
Hi, I took some effort to make an overview of the errors that show when compiling with Java8 (Oracle and Openjdk8) under Debian Jessie. I made 10 runs with multiple compile strategies (mvn compile && mvn package, or mvn package only, ...). None finished OK.

What was strange, was the fact, that the compilation was failing with either 32,33,31 or 8 errors. I'm listing the errors below, sorted by number of occurences in all runs (a summary). If you need, I can provide the logs/more complete error listings, but do not want to spam the forum here.

```
    119 'TestStack.testHttps_TLS_1_0?ServiceResultBad_CommunicationError(code=0x800...'
    119 'TestMaxMessageSize.test_Server_Https_TLS_1_2?ServiceResultBad_Communication...'
    119 'TestMaxMessageSize.test_Server_Https_TLS_1_1?ServiceResultBad_Communication...'
    119 'TestMaxMessageSize.test_Server_Https_TLS_1_0?ServiceResultBad_Communication...'
    119 'TestMaxMessageSize.test_Client_Https_TLS_1_2?ServiceResultBad_Communication...'
    116 'TestMaxMessageSize.test_Client_Https_TLS_1_1?ServiceResultBad_Communication...'
    115 'TestStack.testHttps_TLS_1_1?ServiceResultBad_CommunicationError(code=0x800...'
     69 'TestStack.testOPC_TCP_NONE:215->_setupTest:288->_setupTest:313?ServiceResult'
     69 'TestStack.testOPC_TCP_BASIC256_SIGN_JCE:180->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC256_SIGN_ENCRYPT_JCE:171->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC256_SIGN_ENCRYPT_BC:161->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC256SHA256_SIGN_JCE:209->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC256SHA256_SIGN_ENCRYPT_JCE:200->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC256SHA256_SIGN_ENCRYPT_BC:190->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC128RSA15_SIGN_JCE:151->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC128RSA15_SIGN_ENCRYPT_JCE:142->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC128RSA15_SIGN_ENCRYPT_BC:133->_tearDown:338?NullPointer'
     69 'TestStack.testOPC_TCP_BASIC128RSA15_SIGN_BC:123->_tearDown:338?NullPointer'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_NONE:208->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC256_SIGN_ENCRYPT:118->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC256_SIGN:103->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC256SHA256_SIGN_ENCRYPT:133->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC256SHA256_SIGN:148->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC128RSA15_SIGN_ENCRYPT:88->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Server_OPC_TCP_BASIC128RSA15_SIGN:73->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_NONE:201->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC256_SIGN_ENCRYPT:111->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC256_SIGN:96->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC256SHA256_SIGN_ENCRYPT:126->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC256SHA256_SIGN:141->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC128RSA15_SIGN_ENCRYPT:81->executeTests:232->executeTests:270?ServiceResult'
     69 'TestMaxMessageSize.test_Client_OPC_TCP_BASIC128RSA15_SIGN:66->executeTests:232->executeTests:270?ServiceResult'
     54 'TestMaxMessageSize.test_Client_Https_TLS_1_0?ServiceResultBad_Communication...'
     21 
      1 'TestMaxMessageSize.test_Server_Https_TLS_1_2?ServiceResultBad_Timeout(code...'
      1 'TestMaxMessageSize.test_Server_Https_TLS_1_1?ServiceResultBad_Timeout(code...'
      1 'TestMaxMessageSize.test_Server_Https_TLS_1_0?ServiceResultBad_Timeout(code...'
      1 'TestMaxMessageSize.test_Client_Https_TLS_1_2?ServiceResultBad_Timeout(code...'
      1 'TestMaxMessageSize.test_Client_Https_TLS_1_1?ServiceResultBad_Timeout(code...'
      1 'TestMaxMessageSize.test_Client_Https_TLS_1_0?ServiceResultBad_Timeout(code...'
      1 'TestError>StackTestBench.setUp:84?ServiceResultBad_CommunicationError(code...'
```

Java info
```
# java -version
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.121-b13, mixed mode)

Enabled Protocols: 3
  TLSv1
  TLSv1.1
  TLSv1.2

```
```
# java -version
openjdk version "1.8.0_111"
OpenJDK Runtime Environment (build 1.8.0_111-8u111-b14-2~bpo8+1-b14)
OpenJDK 64-Bit Server VM (build 25.111-b14, mixed mode)

Enabled Protocols: 3
  TLSv1
  TLSv1.1
  TLSv1.2
```