OPCFoundation/UA-Java-Legacy #84: Maven Package Build Failure

teozfrank, 2017-02-07:
I get a build failure when trying to package the repo. I am using java SDK 8 with no need for java 6 support. This is the end of my build log below, any help would be appreciated.

> Results :
> 
> Tests in error: 
>   TestStack.testHttps_TLS_1_0 » ServiceResult Bad_CommunicationError (code=0x800...
> 
> Tests run: 164, Failures: 0, Errors: 1, Skipped: 1
> 
> [INFO] ------------------------------------------------------------------------
> [INFO] BUILD FAILURE
> [INFO] ------------------------------------------------------------------------
> [INFO] Total time: 02:10 min
> [INFO] Finished at: 2017-02-07T12:51:00+00:00
> [INFO] Final Memory: 27M/424M
> [INFO] ------------------------------------------------------------------------
> [ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.19.1:test (default-test) on project opc-ua-stack: There are test failures.
> [ERROR] 
> [ERROR] Please refer to C:\Users\AliveDisplays\Documents\opc-ua\UA-Java\target\surefire-reports for the individual test results.
> [ERROR] -> [Help 1]
> [ERROR] 
> [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
> [ERROR] Re-run Maven using the -X switch to enable full debug logging.
> [ERROR] 
> [ERROR] For more information about the errors and possible solutions, please read the following articles:
> [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
> 
> Process finished with exit code 1

AndreGuerra123, 2017-02-09:
Having the same issue here...I tried both with Maven (v3.3.9) and through Eclipse Neon 2 m2e. 
The debug mode (mvn -x package) reported the following errors:

[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 02:48 min
[INFO] Finished at: 2017-02-09T13:49:03+00:00
[INFO] Final Memory: 26M/322M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.
19.1:test (default-test) on project opc-ua-stack: There are test failures.
[ERROR]
[ERROR] Please refer to E:\OPCTEST\git\UA-Java\target\surefire-reports for the i
ndividual test results.
[ERROR] -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal o
rg.apache.maven.plugins:maven-surefire-plugin:2.19.1:test (default-test) on proj
ect opc-ua-stack: There are test failures.

Please refer to E:\OPCTEST\git\UA-Java\target\surefire-reports for the individua
l test results.
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor
.java:212)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor
.java:153)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor
.java:145)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProje
ct(LifecycleModuleBuilder.java:116)
        at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProje
ct(LifecycleModuleBuilder.java:80)
        at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThre
adedBuilder.build(SingleThreadedBuilder.java:51)
        at org.apache.maven.lifecycle.internal.LifecycleStarter.execute(Lifecycl
eStarter.java:128)
        at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:307)
        at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:193)
        at org.apache.maven.DefaultMaven.execute(DefaultMaven.java:106)
        at org.apache.maven.cli.MavenCli.execute(MavenCli.java:863)
        at org.apache.maven.cli.MavenCli.doMain(MavenCli.java:288)
        at org.apache.maven.cli.MavenCli.main(MavenCli.java:199)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.
java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAcces
sorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced(Laun
cher.java:289)
        at org.codehaus.plexus.classworlds.launcher.Launcher.launch(Launcher.jav
a:229)
        at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode(La
uncher.java:415)
        at org.codehaus.plexus.classworlds.launcher.Launcher.main(Launcher.java:
356)
Caused by: org.apache.maven.plugin.MojoFailureException: There are test failures
.

Please refer to E:\OPCTEST\git\UA-Java\target\surefire-reports for the individua
l test results.
        at org.apache.maven.plugin.surefire.SurefireHelper.reportExecution(Suref
ireHelper.java:91)
        at org.apache.maven.plugin.surefire.SurefirePlugin.handleSummary(Surefir
ePlugin.java:320)
        at org.apache.maven.plugin.surefire.AbstractSurefireMojo.executeAfterPre
conditionsChecked(AbstractSurefireMojo.java:892)
        at org.apache.maven.plugin.surefire.AbstractSurefireMojo.execute(Abstrac
tSurefireMojo.java:755)
        at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo(Default
BuildPluginManager.java:134)
        at org.apache.maven.lifecycle.internal.MojoExecutor.execute(MojoExecutor
.java:207)
        ... 20 more
[ERROR]
[ERROR]
[ERROR] For more information about the errors and possible solutions, please rea
d the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureExc
eption

bjakke, 2017-02-09:
This is (most likely) dublicate of #76. Therefore I would probably close this unless there is anything else failing than testHttps_TLS_1_0? For now options are skipping tests vial -Dmaven.test.skip=true flag or using older java8 version (at least 8u40 works) to compile. 

Hmm.. actually this is kinda complex... I created #85 to improve the situation.

teozfrank, 2017-02-10:
> using older java8 version (at least 8u40 works)

I'm using Java 8u111

> For now options are skipping tests vial -Dmaven.test.skip=true

How would I do that?

teozfrank, 2017-02-10:
Added 

```
         	  <configuration>
			  <skipTests>true</skipTests>
		  </configuration>
```

to this section:

```
      <plugin>
      	<groupId>org.apache.maven.plugins</groupId>
      	<artifactId>maven-surefire-plugin</artifactId>
      	<version>2.19.1</version>
		  <configuration>
			  <skipTests>true</skipTests>
		  </configuration>
      </plugin>`
```

Compiles fine ! :)

```
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 01:21 min
[INFO] Finished at: 2017-02-10T17:32:24+00:00
[INFO] Final Memory: 36M/655M
[INFO] ------------------------------------------------------------------------

Process finished with exit code 0
```

justlikeef, 2017-04-28:
openjdk version "1.8.0_121"
OpenJDK Runtime Environment (IcedTea 3.3.0) (suse-10.2.1-x86_64)
OpenJDK 64-Bit Server VM (build 25.121-b13, mixed mode)

[buildfail.log.txt](https://github.com/OPCFoundation/UA-Java/files/965711/buildfail.log.txt)

kumarmanu4, 2017-06-12:
teozfrank : thank you. the change in POM file did the magic, the package jar was successfully built.

m using JAVA build 1.8.0_73-b02

closed 2017-06-20 by bjakke

sonee123, 2018-03-19:
it's also working for  me... thanks a lot
 <plugin>
      	<groupId>org.apache.maven.plugins</groupId>
      	<artifactId>maven-surefire-plugin</artifactId>
      	<version>2.19.1</version>
		  <configuration>
			  <skipTests>true</skipTests>
		  </configuration>
      </plugin>

parves11, 2018-09-19:
NFO] ------------------------------------------------------------------------
Parvess-MacBook-Air:web-app parves$ mvn compile
[INFO] Scanning for projects...
[INFO] 
[INFO] -----------------------< web-automation:web-app >-----------------------
[INFO] Building web-app 1.0-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ web-app ---
[WARNING] Using platform encoding (UTF-8 actually) to copy filtered resources, i.e. build is platform dependent!
[INFO] skip non existing resourceDirectory /Users/parves/eclipse-workspace/home/web-app/src/main/resources
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ web-app ---
[INFO] Changes detected - recompiling the module!
[WARNING] File encoding has not been set, using platform encoding UTF-8, i.e. build is platform dependent!
[INFO] Compiling 1 source file to /Users/parves/eclipse-workspace/home/web-app/target/classes
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] Source option 5 is no longer supported. Use 6 or later.
[ERROR] Target option 1.5 is no longer supported. Use 1.6 or later.
[INFO] 2 errors 
[INFO] -------------------------------------------------------------
[INFO] -------------------------------------------------

parves11, 2018-09-19:
Source option 5 is no longer supported. Use 6 or later.
[ERROR] Target option 1.5 is no longer supported. Use 1.6 or later.
[INFO] 2 errors 
 whats the mean do you guys know anything about it ?

AttITuder, 2018-12-17:
@teozfrank it's also working for me,thanks a lot.

sjothikr, 2019-05-13:
@parves11

> Source option 5 is no longer supported. Use 6 or later.
> [ERROR] Target option 1.5 is no longer supported. Use 1.6 or later.
> [INFO] 2 errors
> whats the mean do you guys know anything about it ?

check your JDK version. try with JDK1.8, dont go for advance like 1.9,1.10,1.11,1.12. some dependencies removed in advance version.

Vijayalaxmi7, 2019-06-21:
Hi @teozfrank  It is working fine but that it's not providing required output.I don't want to skip my test cases. Please suggest.

teozfrank, 2019-06-21:
Then run it with the tests, if they fail you will have to resolve them yourself or create a new issue to have it resolved. I havn't built the UA Java library in years as I am now using the Prosys OPCUA Java Client SDK.

teozfrank, 2019-06-21:
Or the alternative is to just use the release binaries as they get released -> https://github.com/OPCFoundation/UA-Java-Legacy/releases

vbisht21, 2019-09-11:
Add below dependency in your pom.xml.

<reporting>
		<plugins>
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-surefire-report-plugin</artifactId>
			</plugin>
			<plugin>
				<groupId>org.apache.maven.plugins</groupId>
				<artifactId>maven-jxr-plugin</artifactId>
			</plugin>
		</plugins>
	</reporting>.

commit 71816522 (2017-06-20): Fixes #84 by removing TLS_1.0 Https test

Seems latest Java 8 does not support TLS 1.0 (at least the way it would
need for the current stack to work), therefore removed the test.