OPCFoundation/UA-Java-Legacy #85: Provide pre-built binaries in github releases

bjakke, 2017-02-09:
There are now 2 issues for building the project: #76, #84

It is hard to keep the build runnable on all possible released jdk versions (future and past). For example, newer java versions seem to have deprecated some of the TLS support. I would assume pre-built binaries would be enough for most of the users of the stack. Seems github allows binary attachments to releases so that would be the most practical way to do it currently. Then we could maybe change with the project by default requiring java 6 to build and test as that is how it has been before github era.

teozfrank, 2017-02-10:
That would be great! Please let me know when this goes ahead.

AndreGuerra123, 2017-02-10:
Makes two...

bjakke, 2017-11-20:
Attached build artifacts (binary, javadoc, sources jars) for release 1.03.342.0

closed 2017-11-20 by bjakke