OPCFoundation/UA-Java-Legacy #88: Creating monitored items and change event

karthikvprasad, 2017-03-06:
Hi,

I am trying to write a sample client to monitor the data items in UA server. When a value changes for a particular node, a notification/event has to be triggered. I have succeeded in creating a subscription, but I am stuck at creating monitored items and listening to events. Below is the code I have written, please help me out.

`CreateSubscriptionRequest req = new CreateSubscriptionRequest(); 
CreateSubscriptionResponse subscription = mySession.CreateSubscription(req);
UnsignedInteger subscriptionId = subscription.getSubscriptionId();
UnsignedInteger clientHandle = new UnsignedInteger(clientHandles.getAndIncrement());
ReadValueId ItemToMonitor = new ReadValueId(node, Attributes.Value, null, QualifiedName.NULL);		
MonitoringParameters parameters = new MonitoringParameters(clientHandle, 1000.0, null, new UnsignedInteger(10), true);`

P.S. I am new to Java.

licongchao, 2017-03-10:
same for me. this library doesn't have subscription sample code which is basic usage.

indi7777, 2017-03-17:
I have the same problems. Subscription and MonitorigItems are created, but I do not understand how I can get the ChangeNotifications.
Thanks for any help

jouniaro, 2017-03-17:
The OPC UA Java Stack only implements the communication protocol. That is messaging and security. Session and subscription management are application level issues and are not covered by the plain stack.

The examples that are included are very minimal and provide just very basic functionality related to the Nano Server profile. The Nano Server profile does not require subscriptions, for example.

OPC Foundation actively only focuses on issues in the stack and not on samples, but sample contributions by the community, provided via pull requests, are welcome and will likely be merged.

To get full support for subscriptions you need to use an SDK level product or implement the infrastructure yourself, according to the specification.

dk2129, 2017-03-17:
To get notifications you need to call the "Publish" service and the server will return a notification (if available) in the response.
The Publish service is specified in OPC UA Part 4. It follows the "long polling" pattern, i.e. the server queues the request until it has something to notify. Clients can have multiple Publish requests queued in the Server so that the server always has a channel to return. This requires threading capability in the Client.

As stated by Jouni, this repository is "only" the stack. SDKs provide higher-level support where an application is simply called back whenever a notification arrives.

dk2129, 2017-03-17:
There is some additional text in this post:
https://stackoverflow.com/questions/29522824/opc-ua-subscriptions-and-notifications

indi7777, 2017-03-17:
Okay, thanks for your response. Now I understand the difference between stack and SDK. 
Thanks for your hint with calling the Publish service, but I do not know exactly how i can implement this function. 
Do you have a tip how I can connect the datachangenotification with the Publish service?

Thank you!

dk2129, 2017-03-17:
The Publish service goes to the Session. Any subscription that is created in the session can use such a request to send a notification. You receive the subscription id in the publish response.
Once you get a response you need to send a publish request again and acknowledge the notification passing subscription id(s) and the sequence numbers of received notifications. It is powerful but a bit complicated so in the end, reading the flow in UA Part 4 will help.

indi7777, 2017-03-17:
thank you.
I have understood you so that I must send a Publish request for the subscription and receive a response when the MonitoringItems are changed. After receiving this reply, I need to send a new Publish request for the subscription. Right?
 I have found the function Session.Publish() in the stack.  I will continue to do research at this point. There is also the class DataChangeNotification.

dk2129, 2017-03-17:
As stated - the publish is on the session, the response will include a notification for a specific subscription.

indi7777, 2017-03-17:
Thanks, this is very helpfull.

karthikvprasad, 2017-03-20:
Thank you. This was very helpful. Even I'm doing research on the same. Now I have understood the difference between a SDK and a Stack.

closed 2017-10-17 by jouniaro