OPCFoundation/UA-Java-Legacy #89: Mantis 3735 - handling of unused Variant Type Ids

dk2129, 2017-03-22:
Added rule to UA Part 6:

The Built-in Type Ids 26 through 31 are not currently assigned but may be used in the future. Decoders shall accept these IDs, assume the Value contains a ByteString and pass both onto the application. Encoders shall not use these IDs.

bjakke, 2017-10-17:
This is defined in the UA 1.04 draft, therefore it should not be implemented at least until the specification version 1.04 is released.

bjakke, 2018-11-23:
I'm not exactly sure what should be done in this situation if the Array/ArrayDimension bits are set in the EncodingMask, raised https://opcfoundation-onlineapplications.org/mantis/view.php?id=4478 as I see potentially 2 possible ways it could go (array of ByteStrings vs. a single ByteString, althrough an array of ByteStrings would make more sense)

commit 546642c8 (2018-11-26): Fixed #89, with some assumptions (see comments)

Assumption 1: the text means unknown ids should be treated as-if the
built-in type were ByteString, with all other logic (arrays) working as
is.

Assumption 2: the text is assumed to mean every unknown id, as there are
6 bits of info for the id, therefore 0->63 possible values, so values
26->63 are unknown.
closed 2018-11-30 by bjakke