OPCFoundation/UA-Java-Legacy #93: OPC UA and Java

samber86, 2017-06-10:
Hello all

I am working on a project.

We have one landfill site in germany and from there we have to measure the methane gas concentration and the data is collected by s7 300 plc.

I have to take data from S7 300 PLC and transfer this data to OPC UA server which support java, can you please tell me which OPC UA SERVER will be best to use for this above application and also cheap in cost.

And how can i communicate S7 300 PLC with OPC UA SERVER.

Then i have to programmed the OPC UA CLIENT IN JAVA so please if you have the code of  java  OPC UA client , please share with me

with best regards

Saurabh

jouniaro, 2017-06-12:
Siemens provides an OPC UA server for S7 themselves, but also some other vendors have support for that. 

For OPC UA Java Client application, you can use this Stack implementation, which provides only the low level communication implementation, though - or you can search for a Java SDK, which implements the "toolkit" level and therefore provides a higher level API.

You can search for certified products at https://opcfoundation.org/certified-products

closed 2017-06-12 by jouniaro

samber86, 2017-06-12:
Hello Jo,
Thanks for your reply.

Which OPC UA server used for S7 Can you please tell me and they server also support java language.

And what is Stack Implementation and what is java SDK?  Actually i am new to java

Thanks

regards

jouniaro, 2017-06-12:
You will need to search for S7 products from the market (see my earlier message). We don't have any recommendations.

OPC UA Stack implements the low level communication. OPC UA SDK implements also necessary application level infrastructure and provides a higher level API.

Prosys OPC explains this, for example at https://www.prosysopc.com/opc/

samber86, 2017-06-12:
What about Kepware OPC server which is also for the S7 PLCs.

jouniaro, 2017-06-12:
I am sorry, but this is not the right forum for searching for products or giving any recommendations.

Please, try https://opcfoundation.org/forum instead