OPCFoundation/UA-Java-Legacy #94: Connect to servers with NONE security mode

eoursel, 2017-07-03:
Hi, 

I am just testing the connection with a server which exposes an endpoint without security (NONE). 

The trouble is that the SampleClient is not able to connect with 'no suitable endpoint found'. After some investigations in the source code i found that the call

`	SessionChannel mySession = myClient.createSessionChannel(url);`

does a 

`	tcpEndpoints = EndpointUtil.selectByMessageSecurityMode(tcpEndpoints, MessageSecurityMode.SignAndEncrypt);`

but of course there are NO endpoints with security mode Sign And Encrypt and an exception is thrown without trying the NONE security mode endpoint. 

`SampleClient: Connecting to opc.tcp://169.254.147.99:4845 .. 0    INFO  [main] org.opcfoundation.ua.utils.CryptoUtil     - SecurityProvider initialized from org.bouncycastle.jce.provider.BouncyCastleProvider
1    INFO  [main] org.opcfoundation.ua.utils.CryptoUtil     - Using SecurityProvider BC
30100 INFO  [main] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - /169.254.147.99:4845 Connecting
30168 INFO  [main] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - /169.254.147.99:4845 Connected
30362 INFO  [main] org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp     - -1416681646 Closed
30367 INFO  [TcpConnection/Read] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - /169.254.147.99:4845 Closed (expected)
30368 INFO  [main] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - /169.254.147.99:4845 Closed
Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_UnexpectedError (code=0x80010000, description="No compatible endpoint was found")
	at org.opcfoundation.ua.utils.EndpointUtil.select(EndpointUtil.java:135)
	at org.opcfoundation.ua.utils.EndpointUtil.select(EndpointUtil.java:88)
	at org.opcfoundation.ua.application.Client.createSessionChannel(Client.java:368)
	at org.opcfoundation.ua.application.Client.createSessionChannel(Client.java:345)
	at org.opcfoundation.ua.examples.SampleClient.main(SampleClient.java:109)`

So i would like to understand if I miss something. Is this a bug or something which has been introduced for security? Of course, the connection with a client like UaExpert on this server (Kepware) is ok. 

Regards,

jouniaro, 2017-07-20:
Yes, that does not seem to make sense. There seem to be several 'select' methods in EndpointUtil, which do it a bit differently. I guess the 'select' method that is used here could just skip the filtering based on the MessageSecurityMode.

I believe the reasoning for this implementation comes from the specification that states that the client should always use the best available security mode - although now it limits it even too much.

On the other hand, a better example of making the connection is provided in ClientExample1, which shows how you can select the endpoint yourself, instead of using the "magical" 'select' methods in EndpointUtil.

If you copy that endpoint selection to the SampleClient, you can choose more freely from the available endpoints.

jouniaro, 2017-07-20:
This code will select an insecure endpoint:

`    EndpointDescription[] endpoints = myClient.discoverEndpoints(url);
    // Filter out all but opc.tcp protocol endpoints
    endpoints = selectByProtocol(endpoints, "opc.tcp");
    // Filter out all but Signed & Encrypted endpoints
    endpoints = selectByMessageSecurityMode(endpoints, MessageSecurityMode.None);

    // Choose one endpoint
    if (endpoints.length == 0)
      throw new Exception("The server does not support insecure connections");
    EndpointDescription endpoint = endpoints[0];
    //////////////////////////////////////
    SessionChannel mySession = myClient.createSessionChannel(endpoint);`