OPCFoundation/UA-Java-Legacy #95: Running Sample OPC UA  client  java example

devsylf, 2017-07-31:
Hello,

I am new to OPC UA implementations,and I want to create an OPC UA client.

I have some errors that i did not succeed to solve.

I hope you could help me.

I try to run the SampleClient,that exists on the GitHub of the foundation.

However,I have some policy errors (i think) for which I can’t find the root.

Application myApplication = new Application();
Client myClient = new Client(myApplication);

myApplication.addLocale( ENGLISH );
myApplication.setApplicationName( new LocalizedText(“Java Sample Client”, Locale.ENGLISH) );
myApplication.setProductUri( “urn:JavaSampleClient” );

CertificateUtils.setKeySize(2048); // default = 1024

KeyPair pair = ExampleKeys.getCert(“TestCA”);

That is what i am trying to run.

And the error that it send is:

Exception in thread “main” java.lang.ExceptionInInitializerError
at interfacebd.Connect.main(Connect.java:239)
Caused by: java.lang.RuntimeException: java.io.IOException: keystore password was incorrect
at org.opcfoundation.ua.examples.certs.ExampleKeys.<clinit>(ExampleKeys.java:242)
… 1 more
Caused by: java.io.IOException: keystore password was incorrect
at sun.security.pkcs12.PKCS12KeyStore.engineLoad(PKCS12KeyStore.java:2015)
at java.security.KeyStore.load(KeyStore.java:1445)
at org.opcfoundation.ua.examples.certs.ExampleKeys.<clinit>(ExampleKeys.java:236)
… 1 more
Caused by: java.security.UnrecoverableKeyException: failed to decrypt safe contents entry: java.io.IOException: getSecretKey failed: PBE SecretKeyFactory not available
… 4 more

 

 

I would like to precise that I have already installed all the jar files of policies of java.

Thanks.

jouniaro, 2017-10-17:
Please, try to remove the TestCA.der & .pem from the examples package folder and retry. On the first run, it should create the files and on the second run it tries to read them again.

If you run it with different library setups, you may end up with problems like this: the SunJCE provider is not fully compatible with BC, unfortunately.

But it is also possible that there is some other cause for this that we are not aware of.

bjakke, 2017-12-13:
This is quite clear if you look at the samples i.e. https://github.com/OPCFoundation/UA-Java/blob/master/examples/basic/src/main/java/org/opcfoundation/ua/examples/SampleClient.java. If not, then you need to 1) learn java and 2) learn OPC UA by reading the specification. Prerably (deep) knowledge of both are needed in order to do anything outside of a trivial example with this stack.

mjwSilence, 2017-12-13:
First of all, thank you for answering me，i'm a Java Engineer.haha,lucky. so if i want to connect to the KEPwareServerEX.V4 and just get some data ，i need to 1) import your jar library (by the way,i imported opc-ua-stack-1.03.342.0 .jar,it's right or enough？)    2) create a sampleClient to connect the SERVER ?

mjwSilence, 2017-12-13:
THANK YOU SO MUCH

bjakke, 2017-12-13:
The jars attached to the release are done via 'mvn package', therefore no, it is not enough. Please read the 'Runtime Dependencies' part of the README and dependencies block of https://github.com/OPCFoundation/UA-Java/blob/master/pom.xml. In addition please see the pom.xml of the examples project, in which the SampleClient is https://github.com/OPCFoundation/UA-Java/blob/master/examples/basic/pom.xml.

In general you would typically use a higher level framework/SDK to develop UA applications that just the stack level. Please see my answers on #108 and #109. Anything outside of the samples applications e.g. simple Read service call is going to require lots of effort. It is doable, but since this stack is the bare request/response layer, you _need_ to know OPC UA (i.e. read the part 4 on the service's request/response parameters) in order to do anything.

mjwSilence, 2017-12-13:
i know what do you mean ,but i'm not going to create a new project,i need to connect to the KEPServer in my project which run on tomcat ,so ,can i just import this jar or another more ?  than create a sampleClient 
to connect the KEPServer?

mjwSilence, 2017-12-13:
i mean wo need to see some data from the KEPServer tomorrow , for a simple test ,and than i have time to learn more about UA

mjwSilence, 2017-12-13:
sorry to disturb,buti just find something strange,the  class 'ExampleKeys under the package 'org.opcfoundation.ua.examples.certs' is not packed into the jar 'opc-ua-stack-1.03.342.0 .jar' , Although what you say is not like this， but is this why i can't connect to Server just use the jar ?

mjwSilence, 2017-12-13:
by the way ,i don't need to use it in an encrypted way

bjakke, 2017-12-13:
That package name is used in the samples https://github.com/OPCFoundation/UA-Java/tree/master/examples/basic. It is not part of the stack (i.e. the jar / library). The examples show how your code could look like when you are using the stack (yes, this means that your code should contain e.g. the code in ExampleKeys and SampleClient, of course you need to adapt it to your specific use case). 

The example SampleClient shows a basic Read and Browse service calls, which are the most common ones and easiest to do. At minimum you need to understand the AddressSpace/Node-model + Attributes (NodeId + Value at least)s&References of nodes from the Specification in order to use those. (In addition you might need to configure KEPServer so that it displays some plc tags in the address space at all, please consult on kepservers manual on how to do that). i.e. if you are going to use OPC UA "to see some data from the KEPServer" then you actually do need to know what you are doing, you cannot do it the other way around (at least with this stack i.e. the raw request/response layer).

I recommend that you use a grapchical UA client application, it makes the learning process much easier to understand (and might be anyway need to see e.g. NodeIds), the industry de-facto tool for this has been UaExpert. Alternatively you could try maybe a more advanced toolkit (https://opcfoundation.org/certified-products).

Since the main problem / solution now seems to be a general tutorial on how to use a java library, which is outside the scope of the stack, I will close this issue. If the problem in the first post of this chain persists a new issue can be opened.

closed 2017-12-13 by bjakke

samber86, 2018-02-17:
Hello,

I need the discussion you have in this thread

I want to ask what should i write in the sampleclient  code in place of examplekey

@bjakke is saying the we have used our use case..but what kind of keys available in opc ua

I have read the OPC UA security part ..but i dint understand

can anyone explain

regards

Sam

sc-prayad, 2022-08-22:
Hello,

Can someone help me to communicate opcua client and server.
If possible can you share me source code.
Gmail: ajitkumar.cmrit@gmail.com

Thanks & Regards,
Ajit