OPCFoundation/UA-Java-Legacy #98: Bug testing "TestStackExample.java"

Lutier2017, 2017-09-16:
I was testing "TestStackExample.java"
The test did not work
I got the error :

496  INFO  [main] org.opcfoundation.ua.transport.tcp.io.TcpConnection     - MyComputer/10.53.10.133:6002 Connect failed
org.opcfoundation.ua.common.ServiceResultException: Bad_TcpEndpointUrlInvalid (code=0x80830000, description="Bad_TcpEndpointUrlInvalid (code=0x80830000, description="The Server does not recognize the QueryString specified.")")

==============================

I changed the line 101 :
  From : EndpointDescription[] endpoints = myClient.discoverEndpoints( "opc.tcp://"+hostname+":6002/" ); //51210=Sample Server

  To : EndpointDescription[] endpoints = myClient.discoverEndpoints( "opc.tcp://"+hostname+":6002/UAExample" ); //51210=Sample Server
in the class : TestStackExample
method       : main()

And the test worked fine.
Please document this bug.

jouniaro, 2017-10-16:
OK, apparently the ServerURI provided as a hint for the .NET SampleServer has been incorrect.

bjakke, 2017-11-20:
The test request/responses are no longer present in the resources files for UA 1.03, therefore they are not part of the stack anymore, so this class is also removed.

closed 2017-11-20 by bjakke