OPCFoundation/UA-Java-Legacy #103: ServiceFault: Bad_OutOfMemory (0x80030000) "Not enough memory to complete the operation."

OlegUA, 2017-11-06:
Hi All

I'm trying to connect NI OPC UA server with security policy NONE for test purposes but I'm getting every time the same error (see subj). The full trace below:
Exception in thread "main" ServiceFault: Bad_OutOfMemory (0x80030000) "Not enough memory to complete the operation."
Diagnostic Info: 
	at org.opcfoundation.ua.transport.impl.AsyncResultImpl.waitForResult(AsyncResultImpl.java:243)
	at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.serviceRequest(SecureChannelTcp.java:375)
	at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.serviceRequest(SecureChannelTcp.java:317)
	at org.opcfoundation.ua.application.Client.createSession(Client.java:215)
	at org.opcfoundation.ua.application.Client.createSession(Client.java:153)
	at org.opcfoundation.ua.application.Client.createSession(Client.java:130)
	at org.opcfoundation.ua.application.Client.createSessionChannel(Client.java:395)
	at org.opcfoundation.ua.application.Client.createSessionChannel(Client.java:372)
	at org.opcfoundation.ua.examples.ClientNI.main(ClientNI.java:82)

My test is fully based on ClientExample1.java. I've changed only security endpoint choice and changed ServiceChanned to SessionChannel. I've tried to extend heap memory to 1 Gb via -Xmx1g -Xms1g  parameters but unsuccessfully. 

Could anyone help me sort out what is happened?

Thanks in advance,
Oleg

jouniaro, 2017-11-06:
The error is generated in the server and provided as a response back to the client application. Apparently, the server cannot handle the request for some reason and you will need to contact NI to help you sorting it out.

closed 2017-11-06 by jouniaro

Marconi666, 2024-01-20:
The problem seemed unsolvable, and I searched for it for a long time without finding the reason。