OPCFoundation/UA-Java-Legacy #104: NullPointerException from ExpandedNodeId due to NamespaceTable not having all namespaces

andreas-ibm, 2017-11-21:
This might be a usage problem, but I can't immediately see how to resolve it...

I'm trying to decode an extension object from a server, and it's blowing up in 
`public ExpandedNodeId(UnsignedInteger serverIndex, String namespaceUri, Object value)`
because the namespaceUri is null.

This in turn is caused because 
```java
	public ExpandedNodeId toExpandedNodeId(NodeId nodeId) {
		return new ExpandedNodeId(null, getUri(nodeId.getNamespaceIndex()), nodeId.getValue());
	}
```
Gets null returned by the `getUri` call.  Which is because the `indexUriMap` in `NamespaceTable` only contains 1 element, for namespace 0 (the element I'm reading is in namespace 3).

Is there something I should have invoked previously to ensure the `NamespaceTable` is fully populated?

thanks,
Andreas

bjakke, 2017-11-21:
This is typically handled by higher level SDKs build on top of the Stacks. If you use just the Stack, then you need to do everything manually.

You need to read the NamespaceArray under the Server node, create new NamespaceTable.createFromArray and set it to the EncoderContext (or edit existing one). Then the toExpandedNodeId would work.

Mentioning this just in case, as you will probably run into this next:
In practice the Stack layer supports the Standard models Structures as that is enough to make the request/response layer on top of which SDKs build. You have to do quite much work to support anything else as in addition to the NamespaceTable to be correct, you also need a Serializer for the given Structure or you need to read the DataTypeDictionary system from the server and build one dynamically (or after UA spec 1.04 comes out you could also read it from a new Attribute, if the server supports 1.04). You can look at EncodeableSerializer class for some tips.

closed 2017-11-21 by bjakke

andreas-ibm, 2017-11-21:
Thanks Bjarne, those pointers should give me enough to be dangerous ;-) 
I think we might have a different approach for the Serializer, will see how it goes!