OPCFoundation/UA-Java-Legacy #108: Is there any client sample on how to actually get the measured value from OPC UA Server

irwantantu, 2017-12-07:
Hi,
I am new in the OPC UA, the library is great, but can not find any documentation or setup instruction on how to get read value from measured value or sensors reading from the OPC UA Server..
Can anybody help on this matter?
fyi,
I have ran the SampleClient and it is successfully browse the OPC UA Server, but then how to see the channel, device and tags that is available.. I am using KepServerX UA interface.

Your response is appreciated.. thanks

bjakke, 2017-12-07:
Hi,

This question is outside the scope of the stack, which purpose is to be low level implementation of the request/response channel. But in general you would browse from the root node following forward references and read the the nodes you encounter (or directly if you know their NodeIds). Please read OPC UA Specification Part 3 on the AddressSpace concept and consult on your server provider on how it organizes it.

Typically you would use a graphical client to browse an unknown server, e.g. UaExpert.

closed 2017-12-07 by bjakke

samber86, 2018-02-17:
hello @irwantantu 

Did u become any success in the opc ua client development in java??? i have some question from you

I am also developing OPC UA client in java.. i dont know how to write the code in java..but i have deep knowledge of opc ua.. and i dont know how to use this in java programming

As i am not so much good in programming

irwantantu, 2018-02-19:
Yes… I have a good success of the code. As you are, I have no experience in Java. Let me know how I can help you…

samber86, 2018-02-19:
Can you provide me your email id, so that i can message you there

irwantantu, 2018-02-19:
Irwan.tantu@greensmithenergy.com

himanshumalviya2016, 2018-05-08:
hello @irwantantu  i am also new to OPC UA, need to write client in java to connect server. In java only one example is there in Github. can you please share something so at least i can connect to server and do something farther. i am new to OPC UA need to start from writing client. Help

irwantantu, 2018-05-08:
I was just using the sample available…

Thanks,

[cid:image001.png@01D3B13E.CD95C670]

Irwan Tantu
Irwan.tantu@greensmithenergy.com<mailto:Irwan.tantu@greensmithenergy.com>
Tel +1 844 814-4367