OPCFoundation/UA-Java-Legacy #110: username and password implementation

herleraja, 2017-12-08:
Hi, I was trying to create a user authentication related scenario with username and password. 

//client part of the code
SessionChannel mySession = myClient.createSessionChannel(endpoint);
		mySession.activate("username", "123");

but // what i have to do in server part of code ? 
I have looked into nanoserver example but there is no much explanation for user handling or authentication related information.  

I know the user management is outside the scope of OPCUA (correct me if i'm wrong), so this basic security related requirement is still part of the stack correct?

bjakke, 2017-12-11:
Session service set is part of the authorization (i.e. not authenticating, which is the lower level securechannel on top of which the "session channel" is formed)

Stack handles the SecureChannel Service Set (Specification Part 4, section 5.5), but not the Session Service Set. This is typically handled in a higher level framework, but the NanoServer has a basic example (see onCreate/Activate/CloseSession methods)

herleraja, 2017-12-11:
Hi bjakke,

I got bit confused now.  I'm speaking about the USER authentication at the higher level. If i understood correctly its session service is not handled in stack ? and it is the work of higher level application. correct me if i' wrong.

[image]

I have scenario where i want to authenticate the user and he is authorization to do all the operation (in fact no restriction on authorization part). can you please suggest me how i can do this scenario.

bjakke, 2017-12-11:
Please see my comments on  #109.

In general, this stack handles low level request/response handling system (i.e. the system itself to actually send/receive the messages). In order to to do this, it also handles the SecureChanngel Service Set (Specification Part 4 5.5). Anything else you must code yourself, or use a higher level framwork/SDK that already does it. 

The NanoServer example contains examples on how some of the service sets could be implemented. (As I mentioned above, see see onCreate/Activate/CloseSession methods there).

closed 2017-12-11 by herleraja