OPCFoundation/UA-Java-Legacy #113: SampleClient with NanoServer is throwing Certificate is not valid exception on Https service request

herleraja, 2017-12-15:
I'm running SampleClient with NanoServer and is throwing "Certificate is not valid exception" on Https request but it work fine with tcp. I tried to move the certificate manually from rejected folder to CA. but still it showing same error. 

One more observation the "SampleClientPKI/CA" folder was getting cleared each time when we run sampleClient. So that clear all the existing certificate and start from fresh. 

[image]

Steps: Run NanoServer --> SampleClient (with https)
SampleClient: Connecting to https://127.0.0.1:8443 .. 0    INFO  [main] org.opcfoundation.ua.utils.CryptoUtil     - SecurityProvider initialized from org.bouncycastle.jce.provider.BouncyCastleProvider
0    INFO  [main] org.opcfoundation.ua.utils.CryptoUtil     - Using SecurityProvider BC
14349 INFO  [main] org.opcfoundation.ua.transport.https.HttpsClient     - Enabled protocols in SSL Engine are [SSLv2Hello, TLSv1, TLSv1.1, TLSv1.2]
14363 INFO  [main] org.opcfoundation.ua.transport.https.HttpsClient     - Enabled CipherSuites in SSL Engine are [TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256, TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256, TLS_RSA_WITH_AES_128_CBC_SHA256, TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256, TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256, TLS_DHE_RSA_WITH_AES_128_CBC_SHA256, TLS_DHE_DSS_WITH_AES_128_CBC_SHA256, TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA, TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA, TLS_RSA_WITH_AES_128_CBC_SHA, TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA, TLS_ECDH_RSA_WITH_AES_128_CBC_SHA, TLS_DHE_RSA_WITH_AES_128_CBC_SHA, TLS_DHE_DSS_WITH_AES_128_CBC_SHA, TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256, TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256, TLS_RSA_WITH_AES_128_GCM_SHA256, TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256, TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256, TLS_DHE_RSA_WITH_AES_128_GCM_SHA256, TLS_DHE_DSS_WITH_AES_128_GCM_SHA256, TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA, TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA, SSL_RSA_WITH_3DES_EDE_CBC_SHA, TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA, TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA, SSL_DHE_RSA_WITH_3DES_EDE_CBC_SHA, SSL_DHE_DSS_WITH_3DES_EDE_CBC_SHA, TLS_EMPTY_RENEGOTIATION_INFO_SCSV]
14363 INFO  [main] org.opcfoundation.ua.transport.https.HttpsClient     - Client CipherSuite selection for http://opcfoundation.org/UA/SecurityPolicy#TLS-1-1 is [SSL_RSA_WITH_3DES_EDE_CBC_SHA]
20490 INFO  [Blocking-Work-Executor-1] org.opcfoundation.ua.cert.PkiDirectoryCertificateStore     - Certificate '99326EB5CA5894DFD62D1BC0E17B09C107346870' added to rejected certificates.
Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_CommunicationError (code=0x80050000, description="2147811328, java.security.cert.CertificateException: Certificate is not valid")
	at org.opcfoundation.ua.transport.https.HttpsClientPendingRequest.run(HttpsClientPendingRequest.java:212)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
Caused by: javax.net.ssl.SSLHandshakeException: java.security.cert.CertificateException: Certificate is not valid
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192)
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1949)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:302)
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:296)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1514)
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:216)
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1026)
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:961)
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1062)
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1375)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1403)
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1387)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:553)
	at org.apache.http.conn.ssl.SSLSocketFactory.connectSocket(SSLSocketFactory.java:412)
	at org.apache.http.impl.conn.DefaultClientConnectionOperator.openConnection(DefaultClientConnectionOperator.java:179)
	at org.apache.http.impl.conn.ManagedClientConnectionImpl.open(ManagedClientConnectionImpl.java:328)
	at org.apache.http.impl.client.DefaultRequestDirector.tryConnect(DefaultRequestDirector.java:612)
	at org.apache.http.impl.client.DefaultRequestDirector.execute(DefaultRequestDirector.java:447)
	at org.apache.http.impl.client.AbstractHttpClient.doExecute(AbstractHttpClient.java:884)
	at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:117)
	at org.opcfoundation.ua.transport.https.HttpsClientPendingRequest.run(HttpsClientPendingRequest.java:152)
	... 3 more
Caused by: java.security.cert.CertificateException: Certificate is not valid
	at org.opcfoundation.ua.transport.security.CertValidatorTrustManager.validate(CertValidatorTrustManager.java:71)
	at org.opcfoundation.ua.transport.security.CertValidatorTrustManager.checkServerTrusted(CertValidatorTrustManager.java:95)
	at sun.security.ssl.AbstractTrustManagerWrapper.checkServerTrusted(SSLContextImpl.java:984)
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1496)
	... 19 more

is it having any link to #94. if not then what could be the reason behind this ? Also i read some place that https testing is still going on. So can we use this java stack for https requests for now?

herleraja, 2017-12-15:
Steps: Run NanoServer --> SampleClient (with https)

jouniaro, 2018-01-10:
In general, HTTPS is more complicated and slower to use than the UA TCP connection, so testing has also been prioritised a bit lower.

The main issue with HTTPS certificates is often that they are expected to be signed by a CA certificate, instead of being self-signed. Also in this case, the HTTPS certificates are signed by SampleCA certificate, which is generated by the ExampleKeys. But, in order to trust the certificates signed by this certificate, you must also make the SampleClient to trust the SampleCA certificate.

You can trust the SampleCA by copying it in the SampleClientPKI/CA/certs folder, which appears under the examples project.

I cannot see the folder cleared by myself.

But, getting the connection fixed reveals issue #115. See the workaround mentioned in that issue to get further.

jouniaro, 2018-01-12:
As an alternative, you can disable application authentication altogether, by commenting out the line

`myClient.getApplication().getHttpsSettings().setCertificateValidator(myValidator);`

from SampleClient. In that case, communication will be encrypted with SSL, but application authentication is not used, which is a valid alternative.