OPCFoundation/UA-Java-Legacy #114: Subject class type invalid

laica10, 2018-01-09:
Hey I keep getting this error (0x80010000) which tells me "Subject class type invalid". I'm just using the default ExampleKeys class and still it crashes when creating the http cert, the other 2 work as they should...

I really need a quick fix and I'm not experienced with OPC at all..

Kind regards,
Jarno Gabriel

jouniaro, 2018-01-10:
Which CryptoProvider are you using, i.e. which security libraries have you linked to your project (see the project [readme description](https://github.com/OPCFoundation/UA-Java/blob/master/README.md))?

laica10, 2018-01-10:
I'm using BouncyCastle.. But it only seems to work in the IDE (to generate the certifications) whenever I build the project it doesn't generate the certifications anymore but I'm currently using the certifications generated by the project in the IDE whenever I try to run it on my desktop or the .jar in generally

jouniaro, 2018-01-10:
Any more information. A stack trace, for example?

jouniaro, 2018-01-10:
If this is your issue (https://stackoverflow.com/questions/48157470/opcua-gethttpscert-error) , the stack trace (https://imgur.com/a/BZ5qh) is actually referring to SunJceCryptoProvider. It seems there is an issue with that one, crating the certificate.

Add this line to the beginning of main(), to ensure the Bouncy Castle is available and in use:

`CryptoUtil.setCryptoProvider(new BcCryptoProvider());`

laica10, 2018-01-10:
Fun fact: this is actually my stack voerflow question.. and my stack trace..

jouniaro, 2018-01-10:
Yes, that's what I figured...

laica10, 2018-01-10:
There u go, this was just taken from my console: 
https://imgur.com/a/vLWAA

jouniaro, 2018-01-10:
What's your Java environment?

laica10, 2018-01-10:
I'm using eclipse (IDE) Java 1.8.0_131 JDK

jouniaro, 2018-01-10:
In which platform?

laica10, 2018-01-10:
Mac, also doesn't work on Windows..

jouniaro, 2018-01-10:
A quick search gives this: https://github.com/bcgit/bc-java/issues/156

Which looks like Bouncy Castle is still not properly used. So, please add

Security.addProvider(new BouncyCastleProvider())

to the beginning of your code as well.

I am not sure why it doesn't pick Bouncy Castle for you automatically. Have you perhaps modified the java.security static registration file?

jouniaro, 2018-01-10:
And, don't package the Bouncy Castle libs to your application JAR, if you make one. They are signed and cannot be repackaged.

laica10, 2018-01-10:
I already did the Security one from the beginning I'll try adding the CryptoUtil one too

laica10, 2018-01-10:
Sorry for the late response, I've been really busy.
I'm still getting the same error with the `CryptoUtil.setCryptoProvider(new BcCryptoProvider());` added in the static method of the main class..

jouniaro, 2018-01-11:
Do you still see that DefaultJcaJceHelper in the stack trace?

laica10, 2018-01-11:
Yep sure do, identical stack trace..

jouniaro, 2018-01-12:
Please, enable DEBUG level logging (log4j.properties in resources of examples project for example) and show what your app outputs for available providers. E.g.

DEBUG [main] org.opcfoundation.ua.utils.CryptoUtil     - Providers=[SUN version 1.8, SunRsaSign version 1.8, SunEC version 1.8, SunJSSE version 1.8, SunJCE version 1.8, ...]

OTOH, you can always decide to store the private keys without password in your app or even disable HTTPS; it is an optional transport protocol anyway.

laica10, 2018-01-13:
Still doesn't work it doesn't even show me the debug..
I think I'm just going to deliver the certificates with the app bundled.

jouniaro, 2018-06-19:
Closing the issue which did not get further progress

closed 2018-06-19 by jouniaro