OPCFoundation/UA-Java-Legacy #123: Need help with querying a Node

marinator86, 2018-01-26:
Hello, 

I could use a little help. I'm trying to find a Node by the suffix of its id. Real example:
The id is: "Mario OPC Ethernet Server.EPL:19216830120.C00039:001"
I only know that part "C00039:001". But it is guaranted that it's unique too. Just a showcase anyway.

Can I query for this suffix? I just need a pointer to the docs, or a small sample code. 

Thank you guys in advance!

Mario

jouniaro, 2018-01-30:
OPC UA defines the Query service to help with finding nodes based on different criteria like this. Unfortunately, though, the server applications do not typically support this service, yet. But you can check your server documentation or ask the manufacturer.

If Query is not available, the only thing you can do is to browse the address space until you find the node you are looking for.

closed 2018-01-30 by jouniaro

marinator86, 2018-01-31:
Thank you for your help!