OPCFoundation/UA-Java-Legacy #124: Bad_SessionIdInvalid bug?

OscarLixiaozhou, 2018-01-29:
When I use  client sample("Client Example1")to visit "Nano Server" ,occur a error (Exception in thread "main" ServiceFault: Bad_SessionIdInvalid (0x80250000) "The session id is not valid.").I use another demo server from Unified Automation to test client also ouccur the same error,the error is occured when client try to use read service.(   `ReadResponse res = myChannel.Read(req);`  ).
 
[image]

jouniaro, 2018-01-30:
Looks like this example is incomplete as it does not activate the session before making the read call.

Please, check the SampleClient, which is a better example.

Or you can fix the ClientExample by replacing myChannel with mySession as follows:

`    //////////// TEST-STACK ////////////
    // Create Channel
    // ServiceChannel myChannel = myClient.createServiceChannel(endpoint);
    SessionChannel mySession = myClient.createSessionChannel(url);
    // mySession.activate("username", "123");
    mySession.activate();

    // Create Test Request
    ReadValueId[] nodesToRead = {new ReadValueId(Identifiers.RootFolder, Attributes.BrowseName, null, null)};
    ReadRequest req = new ReadRequest(null, 0.0, TimestampsToReturn.Both, nodesToRead);
    System.out.println("REQUEST: " + req);

    // Invoke service
    ReadResponse res = mySession.Read(req);
    // Print result
    System.out.println("RESPONSE: " + res);
    //////////////////////////////////////

    ///////////// SHUTDOWN /////////////
    // Close channel
    mySession.closeAsync();`

OscarLixiaozhou, 2018-01-30:
I revise the  code as  you metioned,the problem has resovled .Thanks,jouniaro

commit 5404fb78 (2018-03-20): Fixes related to 1.3.343 milestone, updated version to 1.3.343-SNAPSHOT

Fixes #107
Fixes #116
Fixes #119
Fixes #122
Fixes #124

Adds  XmlDecoder.getMatrix
Fixes XmlDecoder.moveToElement used == instead of equals
Fixes XmlDecoder reading Structure with a Structure field
(XmlDecoder.getEncodeable was missing beginField call and
getEncodeableArray was using Class.getName instead of
Class.getSimpleName)
Fixes MultidimensionalArrayUtils.muxArray for situations where the
highest dimension has more than one entry
closed 2018-01-30 by OscarLixiaozhou