OPCFoundation/UA-Java-Legacy #126: OPC UA  Client Development in java

samber86, 2018-02-09:
Hello to all,

I want to design OPC UA client in java and i know to implement OPC UA client we need 3 things
1. application layer
2. SDK
3. Stack

And i am new to the opc ua and also to the java programming

can anyone help me how should i create the client in java and how to use the code mentioned in this website about OPC UA java

best regards

SAM

amitavabasu, 2018-02-09:
More or less, you have to follow these steps:

1)	Download and install java SDK (JDK 1.7 or higher) depending on your environment (Windown, MAC, LINUX etc.). Check: http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html (check license agreement before downloading and using)
2)	Download and Install Git Bash (https://git-scm.com/downloads) based on your environment (Windows, MAC or LINUX) OR Use https://desktop.github.com/ 
3)	Set up Github account to gain access to the source code unless you already have one.
4)	Use git to clone & check out the OPC-UA Java stack (command:>git clone https://github.com/OPCFoundation/UA-Java.git)
5)	Download, Install & setup maven (https://maven.apache.org/download.cgi)
6)	Move to the newly cloned source-code folder & use maven to build OPC-UA Java package (command:>mvn clean package). This will download other java dependencies as well, so you have to be online.
7)	The above command will create “opc-ua-stack-1.03.342.1-SNAPSHOT.jar” in “target” folder inside the project.
8)	Create your Java Application/Project, add the “opc-ua-stack-1.03.342.1-SNAPSHOT.jar” as a dependency along with other dependency you need and want to use. Follow the examples in “org.opcfoundation.ua.examples” package and build your application.
This is how I did it; there may be other options available.

samber86, 2018-02-09:
Hello Amitavabasu,

Thanks a lot for your reply

so i dont have to use the Esclipse java IDE
And what is this Git Bash and Maven

Can you please explain a bit , why not to use Eclipse instead of what are you suggesting

best regards

SAM

amitavabasu, 2018-02-09:
Hi,

You can use any IDE of your choice. Eclipse is perhaps the most popular Java IDE with lot of feature and available plug-ins. You can follow all the steps mentioned above using Eclipse, but it's not mandatory that without Eclipse you can not do it. IDE may make things easier if you have all the right plug-ins installed and know how to use them.

I prefer to clone any github project using git bash, then use eclipse with maven (m2) plugin to create and build Projects including OPC-UA Stack.

Thanks
AB

samber86, 2018-02-09:
Hello...

Thanks for the reply again,

How to use eclipse with maven(m2) plugin ...what is this maven???

can you please send me your project as for the reference...

my email id is: samber.research@gmail.com

samber86, 2018-02-09:
Hello...

One more things how to implement OPC UA Java SDK .

Like 1st i have to create a client and what will be the code for client generation??

then SDK and then Stack

Can you tell

best regards

Saurabh Samber

amitavabasu, 2018-02-09:
See below URL(s) for more information:.

Maven: https://maven.apache.org/
Eclipse m2e: http://www.eclipse.org/m2e/

To answer your last question, you need to look at the example package, after you download the source code. That package contains sample opc-ua-client, opc-ua-server, opc-us-nano-server etc..
in location: <project-location>\examples\basic\src\main\java\org\opcfoundation\ua\examples\

I am an user not a contributor to this project. Just trying to help & increase the number of users to grow the community.

Thanks
AB

samber86, 2018-02-09:
Hello AB;

thanks again for the reply

Which marven i have to install

Binary tar.gz archive | apache-maven-3.5.2-bin.tar.gz | apache-maven-3.5.2-bin.tar.gz.md5 | apache-maven-3.5.2-bin.tar.gz.asc
Binary zip archive | apache-maven-3.5.2-bin.zip | apache-maven-3.5.2-bin.zip.md5 | apache-maven-3.5.2-bin.zip.asc
Source tar.gz archive | apache-maven-3.5.2-src.tar.gz | apache-maven-3.5.2-src.tar.gz.md5 | apache-maven-3.5.2-src.tar.gz.asc
Source zip archive | apache-maven-3.5.2-src.zip | apache-maven-3.5.2-src.zip.md5 | apache-maven-3.5.2-src.zip.asc

samber86, 2018-02-10:
hello AB,

I have cloned the OPC UA Stack program and configure in eclipse.
I have also installed the Maven and plugin in Eclipse oxygen version

When i am opening the UA java stack project in eclipse then i can see the Maven dependencies folder

“opc-ua-stack-1.03.342.1-SNAPSHOT.jar” in “target folder--- this i dint able to find

but i dint able to find any folder consisting

samber86, 2018-02-10:
[image]

see this image how my eclipse project explore look like

thx

amitavabasu, 2018-02-12:
Check your email, I sent a message.

closed 2018-02-15 by jouniaro

amitChechare, 2019-02-17:
Hi.
I am also trying to do the same but getting multiple issues and not getting any other clear documentation in java. Can you please forward your that mail on **amitchechare@gmail.com** it will help me a lot.

thanks in advance.

RamrajRS, 2019-05-21:
Hi,
I am trying to implement OPC client in a software. Could you also pls share the infor here?