OPCFoundation/UA-Java-Legacy #127: OPC UA Stack

samber86, 2018-02-15:
Hello to all,

I have one question, In this UA-Java project , whether it consist of both OPC UA Stack and also OPC UA JAVA SDK.?????????

because when i am using this project it i am getting confused that this only consist of OPC Ua java stack

but for the OPC UA client development i need both stack and sdk 

Please help

regards

Sam

jouniaro, 2018-02-15:
The first sentence of the project README.md says "The OPC Foundation has formally released the Unified Architecture Java Stack and Sample Code to the community."

But, you have correctly understood that developing an application on top of the plain stack is a lot of work and you will need an SDK for that. OPC Foundation is not capable for providing a complete SDK, so you will have to use a commercial SDK that is built on top of the Stack. Look for Prosys OPC UA Java SDK for such a commercial alternative, for example (I am not currently aware of other SDK alternatives built on top of this stack).

closed 2018-02-15 by jouniaro

samber86, 2018-02-15:
But somebody told me that in this OPC UA STACK program "The official OPC Foundation Unified Architecture Java Stack Implementation" also consist of the OPC UA SDK , now i am confused, because in this project the main jar file consist of OPC UA Specifications i.e. OPC UA services and that is SDK

samber86, 2018-02-15:
And i am also using the evaluation version of prosys OPC UA java stack, it consist of sample consol client, but i want to create a Gui based opc ua client.

What will the different when i want to create a GUi opc ua client

please tell

And most of the code in evaluation version is incomplete, and what is the sense of evaluation version

jouniaro, 2018-02-16:
The difference between the Stack and SDK are explained for example by Prosys OPC at https://www.prosysopc.com/opc-ua/ (end of page). This project is the Stack component, which implements the communication protocols, message encoding and security. The SDKs implement application level functionalities and provide higher level APIs.

Usage of Prosys OPC UA Java SDK is out of scope of this project. You will need to contact Prosys OPC about that. 

The examples of this project are very basic and describe how the messaging layer can be used. But it requires the implementation of the OPC UA services on top (which is the job of the SDK or application layer)

samber86, 2018-02-16:
yes Mr. Jouniaro...you are absolutly right...i have already red this link ...anyways thanks again.

I already have the evaluation version of prosys OPC UA Java SDK, i am using it from last 2 weeks and i have read the code and i am also taking help from the pdf of prosys opc ua client tutorial.

But this evaluation version is running from the consol and the OPC UA java client i want to design is GUi based java client.

Can you give me some kind of suggestion that how can i use this code to create my own Gui based client.

And one important question the prosys opc ua java sdk evaluation version consist of necessary code to develop opc ua client.

Or something else needed to be add in the code.

please help

regards

Sam

jouniaro, 2018-02-16:
As said, Prosys product is out of scope of this Github project, hosted by OPC Foundation. Please, ask at the Prosys forum.

But regarding Java development, you need to apply the code to your GUI application design of choice. Neither the stack nor the SDK place any requirements or provide nothing special for the applications in general. They just provide data that you use in your application as you wish. So you will have to learn the basics of Java GUI development first, if you wish to create a GUI application.