OPCFoundation/UA-Java-Legacy #132: structure with optional values

schartlmueller, 2018-03-20:
If a structure contains optional members, the values ​​are not read correctly. In my opinion, the info bits are not interpreted.

UA-Java Version: 1.03.342.0

How the structure is displayed in the UA-Expert:
[image]

When my client reads the structure, the decoder returns the following values:
[image]

If RangeX and RangeY mandatory members, the decoder works.

bjakke, 2018-03-20:
Hi,

Please describe which part/class of this stack is causing problems?

Unions and optional Structures are outside the scope of this stack, at least as of UA 1.03 as the standard information model does not define any, therefore the stack does not need to handle them. Higher-lvl SDKs built on top of Stacks might handle them if they provide support for other information models.

Both optional Structures and Unions are encoding specific optimizations added in UA 1.03, earlier clients do not understand them at all. In practice it is a more compact way to encode null values (a Union is a Structure with only one field having non-null value). It also provides a way to encode null values for types that do not define a null encoding in Part 6 of the specification, e.g. integer types. Therefore it is only related to the Serializers of such Structures. Serializers can be added via:

`EncodeableSerializer.getInstance().addSerializer(...)`

Typically higher-lvl SDKs generate serializers from information models and add them via that method.

schartlmueller, 2018-03-20:
I think the problem is in the BinaryDecoder. When my Serializer calls the method getDouble("X") it already returns the wrong value. 

If you look at the HeapByteBuffer with the debugger, you see that there are info bits at the beginning and still the method getDouble("X") begins to read the value from position 0 onwards.

[image]

bjakke, 2018-03-20:
Can you show the code of the ProfilePointSerializer?

But in practice you (your serializer code) needs to know (when making the serializer) that does the Structure (or if it inherits from any) have any optional fields (or is it a Union) and more specifically which fields can be optional. If it does, then you need to read a decoder.getUInt32() as the first "field", note that it is not an actual field but instead meta information about which optional fields are serialized. Then for optional Structures the first bit i.e. position 0 tells if the first optional field is present, if not then you must skip it completely, then the next position (1) does tell if the next optional is present etc. (note that there might be non-optional fields between them and the first field might not be optional, i.e. you need to completely know the Structure). For Unions taking the whole UInt32 value, 0 means null, and others is the order of the field which is present (others are null), you must only get this field from the decoder. As an additional note, this "meta" field is only present if there are any optional fields (or the Structure is Union), otherwise for normal Structures the first field starts the encoded form.

schartlmueller, 2018-03-20:
Thanks for the hint. I was not aware that the bitmask needs to be interpreted by the serializer.

closed 2018-03-20 by schartlmueller