OPCFoundation/UA-Java-Legacy #134: Serializer calcEncodeable & putEncodeable

schartlmueller, 2018-03-28:
Hi!

Why does the stack call the calcEncodeable method three times when writing a structure? Is there a difference between calcEncodeable and putEncodeable?
  
`public class ProfilePointSerializer extends AbstractSerializer {

	public static final ExpandedNodeId ID = new ExpandedNodeId("http://yourorganisation.org/profile/",
			Integer.valueOf(3002));
	public static final ExpandedNodeId BINARY = new ExpandedNodeId("http://yourorganisation.org/profile/",
			Integer.valueOf(5001));

	private static final String X = "x";
	private static final String Y = "y";
	private static final String RANGE_X = "rangeX";
	private static final String RANGE_Y = "rangeY";

	private DataTypeConverter conv = null;

	/**
	 * @param clazz
	 * @param binaryId
	 * @param xmlId
	 */
	public ProfilePointSerializer(DataTypeConverter conv) {
		super(ProfilePoint.class, BINARY, null);
		this.conv = conv;
	}

	/*
	 * (non-Javadoc)
	 * 
	 * @see org.opcfoundation.ua.encoding.utils.AbstractSerializer#getEncodeable(org.opcfoundation.ua.
	 * encoding.IDecoder)
	 */
	@Override
	public IEncodeable getEncodeable(IDecoder decoder) throws DecodingException {
		System.out.println("getEncodeable");
		int bitmask = decoder.getUInt32("").intValue();
		ProfilePoint result = new ProfilePoint();
		result.setX(decoder.getDouble(X));
		result.setY(decoder.getDouble(Y));
		result.setRangeX(isOptionalMemberAvailable(bitmask, 0)
				? conv.convertToTargetType(decoder.getEncodeable(RANGE_X, Range.class), Codomains.class)
				: null);
		result.setRangeY(isOptionalMemberAvailable(bitmask, 1)
				? conv.convertToTargetType(decoder.getEncodeable(RANGE_Y, Range.class), Codomains.class)
				: null);

		return result;
	}

	// TODO: export to Util-Class
	private static boolean isOptionalMemberAvailable(int bitmask, int position) {
		return ((bitmask >> position) & 1) == 1;
	}

	// TODO: export to Util-Class
	private static int setOptionalMemberAvailable(int bitmask, int position) {
		return bitmask | (1 << position);
	}
	
	/*
	 * (non-Javadoc)
	 * 
	 * @see
	 * org.opcfoundation.ua.encoding.utils.AbstractSerializer#calcEncodeable(org.opcfoundation.ua.
	 * encoding.IEncodeable, org.opcfoundation.ua.encoding.IEncoder)
	 */
	@Override
	public void calcEncodeable(IEncodeable encodeable, IEncoder calculator) throws EncodingException {
		System.out.println("calcEncodeable");
		createEncodeable(encodeable, calculator);
	}

	/*
	 * (non-Javadoc)
	 * 
	 * @see org.opcfoundation.ua.encoding.utils.AbstractSerializer#putEncodeable(org.opcfoundation.ua.
	 * encoding.IEncodeable, org.opcfoundation.ua.encoding.IEncoder)
	 */
	@Override
	public void putEncodeable(IEncodeable encodeable, IEncoder encoder) throws EncodingException {
		System.out.println("putEncodeable");
		createEncodeable(encodeable, encoder);
	}

	/**
	 * @param encodeable
	 * @param encoder
	 * @throws EncodingException
	 */
	private void createEncodeable(IEncodeable encodeable, IEncoder encoder) throws EncodingException {
		ProfilePoint obj = (ProfilePoint) encodeable;
		int bitMask = 0;

		if (obj != null) {
			if (obj.getRangeX() != null) {
				bitMask = setOptionalMemberAvailable(bitMask, 0);
			}
			if (obj.getRangeY() != null) {
				bitMask = setOptionalMemberAvailable(bitMask, 1);
			}
		}

		encoder.putUInt32("", UnsignedInteger.getFromBits(bitMask));
		encoder.putDouble(X, obj == null ? null : obj.getX());
		encoder.putDouble(Y, obj == null ? null : obj.getY());
		encoder.put(RANGE_X, obj == null ? null : conv.convertToTargetType(obj.getRangeX(), Range.class), Range.class);
		encoder.put(RANGE_Y, obj == null ? null : conv.convertToTargetType(obj.getRangeY(), Range.class), Range.class);
	}`

bjakke, 2018-03-29:
Hi,

Thanks for the issue. If possible, please list all (3) stacktraces that call the calcEncodeable in addition to the code you are using for Write.

In general I guess the intention of the original stack developers has been that the calcEncodeable method is used to calculate the size of the encoded form in bytes and the putEncodeable is then used to actually create the encoded form. There are limits on how large messages can be sent back, so in some sense checking the size first might sound a good idea. Also the base for the original code might have been other stacks, whose programming language might not allow dynamic arrays (like ArrayList in java), which might have influenced the design.

The current architecture would need to change quite heavily in order to improve this. Currently the entire message is encoded as a stream of bytes, so e.g. when putting a Structure to BinaryEncoder we need to know the size before starting to put the structure fields, so it uses EncoderCalc (which uses the calcEncodeable methods in serializers). Given that the entire ServiceRequests are also Structures in practice, this effect will multiply few times.

schartlmueller, 2018-03-29:
Stracktrace:
#1: calcEncodeable
OpcClientTest [Java Application]	
	com.test.datamodel.client.remote.OpcClientTest at localhost:33263	
		Thread [main] (Suspended (breakpoint at line 108 in ProfilePointSerializer))	
			ProfilePointSerializer.createEncodeable(IEncodeable, IEncoder) line: 108	
			ProfilePointSerializer.calcEncodeable(IEncodeable, IEncoder) line: 56	
			ProfilePointSerializer(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putEncodeable(String, IEncodeable) line: 1623	
			EncoderCalc.putStructure(String, Structure) line: 1108	
			EncoderCalc.putStructureArray(String, Structure[]) line: 1127	
			EncoderCalc.putArray(String, int, Object) line: 1591	
			EncoderCalc.putVariant(String, Variant) line: 1301	
			EncoderCalc.putDataValue(String, DataValue) line: 1228	
			EncodeableSerializer$135.calcEncodeable(IEncodeable, IEncoder) line: 3729	
			EncodeableSerializer$135(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putEncodeableArray(String, Class<IEncodeable>, Object) line: 1611	
			EncodeableSerializer$136.calcEncodeable(IEncodeable, IEncoder) line: 3754	
			EncodeableSerializer$136(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putMessage(IEncodeable) line: 1681	
			TcpConnection.sendRequest(ServiceRequest, int, int) line: 1094	
			SecureChannelTcp.serviceRequest(ServiceRequest, long) line: 342	
			SecureChannelTcp.serviceRequest(ServiceRequest) line: 315	
			ClientSession(InternalClient).write(WriteRequest) line: 999	
			NodeManager.write(ClientSession, WriteRequest) line: 319	
			ProfileManager.write2(ClientSession, NodeId[], UnsignedInteger[], String[], DataValue[], boolean) line: 2217	
			ProfileManager.write1(ClientSession, NodeId[], UnsignedInteger[], String[], DataValue[], boolean) line: 2159	
			UAClient.write1(ClientSession, NodeId, UnsignedInteger, String, DataValue) line: 2230	
			OpcClient.writeActPoints(String, NodeId) line: 708	
			OpcClientTest.main(String[]) line: 31	
		Daemon Thread [KeepAliveTimer] (Running)	
		Daemon Thread [UA Timer] (Running)	
		Daemon Thread [TcpConnection/Read] (Running)	
		Daemon Thread [Blocking-Work-Executor-1] (Running)	
	/usr/lib/jvm/jdk1.9/bin/java (29.03.2018, 14:22:48)	

#2: calcEncodeable
OpcClientTest [Java Application]	
	com.test.datamodel.client.remote.OpcClientTest at localhost:33263	
		Thread [main] (Suspended (breakpoint at line 108 in ProfilePointSerializer))	
			ProfilePointSerializer.createEncodeable(IEncodeable, IEncoder) line: 108	
			ProfilePointSerializer.calcEncodeable(IEncodeable, IEncoder) line: 56	
			ProfilePointSerializer(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putEncodeable(String, IEncodeable) line: 1623	
			EncoderCalc.putStructure(String, Structure) line: 1108	
			EncoderCalc.putStructureArray(String, Structure[]) line: 1127	
			EncoderCalc.putArray(String, int, Object) line: 1591	
			EncoderCalc.putVariant(String, Variant) line: 1301	
			EncoderCalc.putDataValue(String, DataValue) line: 1228	
			EncodeableSerializer$135.calcEncodeable(IEncodeable, IEncoder) line: 3729	
			EncodeableSerializer$135(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putEncodeableArray(String, Class<IEncodeable>, Object) line: 1611	
			EncodeableSerializer$136.calcEncodeable(IEncodeable, IEncoder) line: 3754	
			EncodeableSerializer$136(AbstractSerializer).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 102	
			EncodeableSerializer(SerializerComposition).calcEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 98	
			EncoderCalc.putMessage(IEncodeable) line: 1681	
			TcpConnection.encodeMessage(ChunkFactory, int, IEncodeable) line: 1296	
			TcpConnection.sendRequest(ServiceRequest, int, int) line: 1110	
			SecureChannelTcp.serviceRequest(ServiceRequest, long) line: 342	
			SecureChannelTcp.serviceRequest(ServiceRequest) line: 315	
			ClientSession(InternalClient).write(WriteRequest) line: 999	
			NodeManager.write(ClientSession, WriteRequest) line: 319	
			ProfileManager.write2(ClientSession, NodeId[], UnsignedInteger[], String[], DataValue[], boolean) line: 2217	
			ProfileManager.write1(ClientSession, NodeId[], UnsignedInteger[], String[], DataValue[], boolean) line: 2159	
			UAClient.write1(ClientSession, NodeId, UnsignedInteger, String, DataValue) line: 2230	
[... 60 lines cut ...]
			BinaryEncoder.putDataValue(String, DataValue) line: 2086	
			EncodeableSerializer$135.putEncodeable(IEncodeable, IEncoder) line: 3736	
			EncodeableSerializer$135(AbstractSerializer).putEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 113	
			EncodeableSerializer(SerializerComposition).putEncodeable(Class<IEncodeable>, IEncodeable, IEncoder) line: 90	
[... cut ...]
			ProfileManager.write1(ClientSession, NodeId[], UnsignedInteger[], String[], DataValue[], boolean) line: 2159	
			UAClient.write1(ClientSession, NodeId, UnsignedInteger, String, DataValue) line: 2230	
			OpcClient.writeActPoints(String, NodeId) line: 708	
			OpcClientTest.main(String[]) line: 31	
		Daemon Thread [KeepAliveTimer] (Running)	
		Daemon Thread [UA Timer] (Running)	
		Daemon Thread [TcpConnection/Read] (Running)	
		Daemon Thread [Blocking-Work-Executor-1] (Running)	
	/usr/lib/jvm/jdk1.9/bin/java (29.03.2018, 14:22:48)