OPCFoundation/UA-Java-Legacy #139: TranslateBrowsePathsToNodeIds does not work

viclen, 2018-05-10:
I have the opc ua server running, from the server i can print the default folders, e.g. "Root" or "Objects". But i cant find any other nodes below them.  Im trying to translate the browse path with TranslateBrowsePathsToNodeIds but always return Bad_NoMatch.

`String[] pathParts = "/Objects/3:ClpAgricolaAlvorada/3:DataBlocksGlobal/3:DadosSetpointHorimetros/".split("/");
    RelativePathElement[] element = new RelativePathElement[pathParts.length];
    for(int i=0; i<pathParts.length; i++){
        element[i] = new RelativePathElement(new NodeId(0,i+1), false, true, new QualifiedName(pathParts[i]));
        System.out.println("Adicionado: "+element[i]);
    }
    RelativePath rp = new RelativePath(element);
    System.out.println(rp);
    BrowsePath path = new BrowsePath(Identifiers.RootFolder, rp);
    System.out.println(path);
    TranslateBrowsePathsToNodeIdsResponse tr = mySession.TranslateBrowsePathsToNodeIds(null, path);
    
    System.out.println(tr);`

Ps. Already tried without the ":" in browse paths.

viclen, 2018-05-10:
Is there anything i can do to map all nodes in my plc?

htahvanainen, 2018-05-11:
Hi, 

As an immediate issue, you're not creating the RelativePathElement correctly and this is the reason why the TranslateBrowsePathsToNodeIds service call will not function properly. Please see below two points about what to correct:

- When creating the RelativePathElement, you'll need to supply the type of reference to follow from the current node. Instead, you're now putting "new NodeId(0,i+1)" as a parameter which doesn't mean anything sensible. If you want to include all reference types, you may give "null" as a parameter. You may also put some actual reference type as a parameter, such as Identifiers.HierarchicalReferences.

- As a second point for the RelativePathElement, you are now creating all of the targetName parameters in the namespace 0. This is the parameter "new QualifiedName(pathParts[i])". Instead, the QualifiedName needs to be created with the correct namespace index. In the provided example case, most of the browse names reside in namespace index 3. 

When you correct these points, the TranslateBrowsePathsToNodeIds call should work. 

However, the more pressing issue is that you should not be using the TranslateBrowsePathsToNodeIds service to "map all nodes in my plc". In OPC UA, there's a Browse service which is used for this purpose. So, please see the Browse service which is much more fitting for your use case.

closed 2018-05-11 by htahvanainen