OPCFoundation/UA-Java-Legacy #140: Browsing doesn't work.

Pllvcn83, 2018-05-11:
when I browse the server I see the first node but non the second. i think there is a bug in java.

the response:

 BrowseResponse
  ResponseHeader=ResponseHeader
    Timestamp=05/11/18 14:19:21.2503015 GMT
    RequestHandle=0
    ServiceResult=GOOD (0x00000000) ""
    ServiceDiagnostics=Diagnostic Info: 
      
    StringTable=class java.lang.String[0]
    AdditionalHeader=null
  Results=class org.opcfoundation.ua.core.BrowseResult[1]
    [0]=BrowseResult
    StatusCode=GOOD (0x00000000) ""
    ContinuationPoint=null
    References=class org.opcfoundation.ua.core.ReferenceDescription[2]
      [0]=ReferenceDescription
      ReferenceTypeId=i=35
      IsForward=true
      NodeId=nsu=http%3A%2F%2Fopcfoundation.org%2FUA%2F;i=2253
      BrowseName=Server
      DisplayName=Server
      NodeClass=NodeClass
        value=1
        name=Object
        ordinal=1
      TypeDefinition=nsu=http%3A%2F%2Fopcfoundation.org%2FUA%2F;i=2004
      [1]=ReferenceDescription
      ReferenceTypeId=i=35
      IsForward=true
      NodeId=ns=2;s=WRT
      BrowseName=2:WRT
      DisplayName=WRT
      NodeClass=NodeClass
        value=1
        name=Object
        ordinal=1
      TypeDefinition=nsu=http%3A%2F%2Fopcfoundation.org%2FUA%2F;i=58
  DiagnosticInfos=class org.opcfoundation.ua.builtintypes.DiagnosticInfo[0]

2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - i = 0 x= 0
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseName   Server
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - NodeId       nsu=http%3A%2F%2Fopcfoundation.org%2FUA%2F;i=2253
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - DisplayName  Server
2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseResponse
  ResponseHeader=ResponseHeader
    Timestamp=05/11/18 14:19:21.2522941 GMT
    RequestHandle=0
    ServiceResult=GOOD (0x00000000) ""
    ServiceDiagnostics=Diagnostic Info: 
      
    StringTable=class java.lang.String[0]
    AdditionalHeader=null
  Results=class org.opcfoundation.ua.core.BrowseResult[1]
    [0]=BrowseResult
    StatusCode=GOOD (0x00000000) ""
    ContinuationPoint=null
    References=class org.opcfoundation.ua.core.ReferenceDescription[0]
  DiagnosticInfos=class org.opcfoundation.ua.builtintypes.DiagnosticInfo[0]

when i call the secon time the method browse I see:

2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - i = 0 x= 0
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseName   Server
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - NodeId       nsu=http%3A%2F%2Fopcfoundation.org%2FUA%2F;i=2253
2018-05-11 16:19:21,251 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - DisplayName  Server
2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseResponse
  ResponseHeader=ResponseHeader
    Timestamp=05/11/18 14:19:21.2522941 GMT
    RequestHandle=0
    ServiceResult=GOOD (0x00000000) ""
    ServiceDiagnostics=Diagnostic Info: 
      
    StringTable=class java.lang.String[0]
    AdditionalHeader=null
  Results=class org.opcfoundation.ua.core.BrowseResult[1]
    [0]=BrowseResult
    StatusCode=GOOD (0x00000000) ""
    ContinuationPoint=null
    References=class org.opcfoundation.ua.core.ReferenceDescription[0]
  DiagnosticInfos=class org.opcfoundation.ua.builtintypes.DiagnosticInfo[0]

2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - i = 0 x= 1
2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseName   2:WRT
2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - NodeId       ns=2;s=WRT
2018-05-11 16:19:21,252 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - DisplayName  WRT
2018-05-11 16:19:21,253 [Component Resolve Thread (Bundle 6)] DEBUG c.a.o.u.ClientService - BrowseResponse
  ResponseHeader=ResponseHeader
    Timestamp=05/11/18 14:19:21.2532972 GMT
    RequestHandle=0
    ServiceResult=GOOD (0x00000000) ""
    ServiceDiagnostics=Diagnostic Info: 
      
    StringTable=class java.lang.String[0]
    AdditionalHeader=null
  Results=class org.opcfoundation.ua.core.BrowseResult[1]
    [0]=BrowseResult
    StatusCode=GOOD (0x00000000) ""
    ContinuationPoint=null
    References=class org.opcfoundation.ua.core.ReferenceDescription[0]
  DiagnosticInfos=class org.opcfoundation.ua.builtintypes.DiagnosticInfo[0]

this is wrong. Because when I use a C# client i can see the next node.
My code here:

   public void browse(NodeId nodeId) throws ServiceFaultException, ServiceResultException {
        BrowseDescription browse = new BrowseDescription();
        browse.setNodeId(nodeId);
        browse.setBrowseDirection(BrowseDirection.Forward);
        browse.setIncludeSubtypes(true);
        browse.setNodeClassMask(NodeClass.Object, NodeClass.Variable, NodeClass.View);
        browse.setResultMask(BrowseResultMask.All);
        BrowseResponse browseResponse = null;
        int k = 1;
        try {
            browseResponse = mySession.Browse(null, null, null, browse);
            logger.debug(browseResponse.toString());

        } catch (ServiceResultException e) {

            logger.error("Exception", e);
        }

        
        
        BrowseResult[] browseResult = browseResponse.getResults();
     
        for (int i = 0; i < browseResult.length; i++) {
            ReferenceDescription[] referenceDescription = browseResult[i].getReferences();

            for (int x = 0; x < referenceDescription.length; x++) {
           logger.debug("i = " + i + " x= " + x);
           logger.debug("BrowseName   " + referenceDescription[x].getBrowseName());
           logger.debug("NodeId       " + referenceDescription[x].getNodeId());
           logger.debug("DisplayName  " + referenceDescription[x].getDisplayName());

               
                    BrowseDescription browse1 = new BrowseDescription();
                    browse1.setNodeId( referenceDescription[x].getNodeId().ID);
                    browse1.setBrowseDirection(BrowseDirection.Forward);
                    browse1.setIncludeSubtypes(true);
                    browse1.setReferenceTypeId(referenceDescription[x].getTypeDefinition().ID);
                    browse1.setNodeClassMask(NodeClass.Object, NodeClass.Variable, NodeClass.View, NodeClass.Method, NodeClass.ObjectType, NodeClass.VariableType,   NodeClass.ReferenceType , NodeClass.DataType);
                    browse1.setResultMask(BrowseResultMask.ALL);
                    BrowseResponse browseResponse1 = null;
                    try {
                        browseResponse1 = mySession.Browse(null, null, null, browse1);
                        
                        if (browseResponse1 != null) {
                            logger.debug(browseResponse1.toString());
                        }
                    } catch (ServiceResultException e) {

                        logger.error("Exception", e);
                    }

              

            }

        }
    }

jouniaro, 2018-05-11:
As mentioned in https://stackoverflow.com/questions/50292359/opc-ua-fondation-java-browsing, you are using wrong parameters for the second browse. Use

browse1.setNodeId( referenceDescription[x].getNodeId());

and don't call 'browse1.setReferenceTypeId()'

closed 2018-05-11 by jouniaro

Pllvcn83, 2018-05-11:
How to wrote there.. I removed 'browse1.setReferenceTypeId()' and i used browse1.setNodeId( referenceDescription[x].getNodeId().ID); and it doesent work. maybe beacuse  donesn't return NodeID?!?

jouniaro, 2018-05-11:
Don't include the '.ID'

Pllvcn83, 2018-05-11:
I understood what you mean. But if i remove '.ID' it doesen' t  build because referenceDescription[x].getNodeId() returns an object of ExpandedNodeId.

Pllvcn83, 2018-05-11:
is there a way to turn an ExpandedNodeId in a NodeId?

jouniaro, 2018-05-11:
Oh yes, this is a complication. Unfortunately, OPC UA defines these two different types. You need a NamespaceTable to do the conversion properly. For basic nodes, you can use 'NamespaceTable.getDefaultInstance()'.  'toNodeId()' is then the method you need.