OPCFoundation/UA-Java-Legacy #141: Security settings and user authentication

mmattia17, 2018-05-14:
I'm trying to establish a secure connection between my server and a Java Client In node js defined as in the follow:

```javascript
var userManager = {
    isValidUser: function (userName, password) {
        if (userName === "user1" && password === "password1") {
            return true;
        }
        if (userName === "user2" && password === "password2") {
            return true;
        }
        return false;
    }
};
var server_certificate_file = constructFilename("certificates/server_cert_2048.pem");
var server_certificate_privatekey_file = constructFilename("certificates/server_key_2048.pem");

 var server = new opcua.OPCUAServer({
    port: port, // the port of the listening socket of the server
    resourcePath: resourcePath, // this path will be added to the endpoint resource name

      securityPolicies: [
          SecurityPolicy.Basic128Rsa15,
          SecurityPolicy.Basic256
      ],
      securityModes: [
          MessageSecurityMode.SIGN,
          MessageSecurityMode.SIGNANDENCRYPT
      ],
      certificateFile: server_certificate_file,
      privateKeyFile: server_certificate_privatekey_file,
      allowAnonymous : false,
      userManager: userManager,

    buildInfo : {
      productName: "MillingMachineServer",
      buildNumber: "1",
      buildDate: new Date(2018,4,10)
    }
  });

```

The server works fine, if I use it with a simple nodejs client, but if I'm trying to connect to it with a java client as reported below, 

```java

		Application myClientApplication = new Application();

	    KeyPair myClientApplicationInstanceCertificate = null;

	    // Create Client Application Instance Certificate
	    {
	      String certificateCommonName = "UA Sample Client";
	      System.out.println("Generating new Certificate for Client using CN: " + certificateCommonName);

	      String applicationUri = myClientApplication.getApplicationUri();
	      String organisation = "Sample Organisation";
	      int validityTime = 365;
	      myClientApplicationInstanceCertificate = CertificateUtils
	          .createApplicationInstanceCertificate(certificateCommonName, organisation, applicationUri, validityTime);
	    }

	    // Create Client
	    Client myClient = new Client(myClientApplication);
	   myClientApplication.addApplicationInstanceCertificate(myClientApplicationInstanceCertificate);
	    //////////////////////////////////////

		// Create the client using information provided by the created certificate

	    /////////// DISCOVER ENDPOINT ////////
	    // Discover server's endpoints, and choose one
	    EndpointDescription[] endpoints = myClient.discoverEndpoints(url);//51210=Sample
	                                                                                                        // Server
	    // Filter out all but opc.tcp protocol endpoints
	    endpoints = EndpointUtil.selectByProtocol(endpoints, "opc.tcp");
	    // Filter out all but Signed & Encrypted endpoints
	    
	    endpoints = EndpointUtil.selectByMessageSecurityMode(endpoints,MessageSecurityMode.SignAndEncrypt);
	    
	    // Filter out all but Basic128 cryption endpoints
	    endpoints = EndpointUtil.selectBySecurityPolicy(endpoints, SecurityPolicy.BASIC256);
	    
	    // Sort endpoints by security level. The lowest level at the beginning, the highest at the end
	    // of the array
	    endpoints = EndpointUtil.sortBySecurityLevel(endpoints);
	    // Choose one endpoint
	    EndpointDescription endpoint = endpoints[endpoints.length - 1];
	    //////////////////////////////////////

		
		
		///////////// EXECUTE //////////////
		SessionChannel mySession = myClient.createSessionChannel(endpoint);
		mySession.activate("user1","password1");

```
I have this kind of exception due to the fact that my Session can't retrieve a the certificate from my server.

```java

java.lang.NullPointerException
	at org.opcfoundation.ua.utils.EndpointUtil.createUserNameIdentityToken(EndpointUtil.java:353)
	at org.opcfoundation.ua.application.SessionChannel.activate(SessionChannel.java:115)
	at MappingTool.main(MappingTool.java:232)

```
I've fixed this bug modifying the createSessionChannel in the Client.java of the package org.opcfoundation.ua.application

```java
public SessionChannel createSessionChannel(String connectUrl, EndpointDescription endpoint) throws ServiceResultException 
	{
		// Create service channel
		SecureChannel channel = createSecureChannel(connectUrl, endpoint);
		try {
			// Create session
			Session session = createSession(channel);

			//HOW I'VE FIXED
			session.getEndpoint().setServerCertificate(endpoint.getServerCertificate());
			//END FIXING

			// Create session channel
			SessionChannel sessionChannel = session.createSessionChannel(channel, this);
			return sessionChannel;
		} catch (ServiceResultException se) {
			channel.closeAsync();
			throw se;
		}
	}
```

Am I wrong using the library to instantiate a secure connection? How can I solve this problem?

bjakke, 2018-05-17:
Hi,

The following should work without the need to modify code (pretty much does the same, but through different overloads):

Client client = ...
session = client.createSession(secureChannel, ...
session.getEndpoint().setServerCertificate(ByteString.valueOf(session.getServerCertificate().getEncoded())
channel = session.createSessionChannel(...)
(use the channel for request/responses)

Mainly the reasons are that the Stack's legacy code is very old, and later versions of the specification allows setting some parameters of EndpointDescription to null in CreateSessionResponse (since it was redundant and already obtained from GetEndpoints call), which breaks earlier clients made to older versions.

mmattia17, 2018-05-17:
Thank you. This solution works fine.

closed 2018-05-17 by mmattia17