OPCFoundation/UA-Java-Legacy #148: Is it possible to use the current up-to-date stack in pub/sub project ?

hkarabakla, 2018-06-05:
Hi,
I want to implement an opcua pub/sub application with Java and I couldn't understand whether the current up-to-date stack allows me to build on it ? If it doesn't, could you give me tips about how to do that on my own and is there any plan to support pub/sub in the near feature ?

bjakke, 2018-06-07:
Do you mean the new UA 1.04 specification Part 14 PubSub? (answering assuming that)

At the moment it is not possible even theoretically, as this stack is at UA 1.03. However it should be noted that this stack handles only connection, certs and raw request/responses for Part 4 ServiceRequests (technically also Part 6 for encoding those). And pubsub is a completely new "addon" Part 14, which for most parts wont be handled by this stack (must be handled on application/SDK levels).

In general the following is needed at minimum:
- 1.04 support for generated parts of the stack #136 
- JSON encoding support (no issue key yet, also the above is much higher priority)
- UADP support maybe

After that it should be possible (however it should be noted that in general it most likely requires quite **a lot** of work) as the standard namespace resources includes pubsub Structure and datatypes and pubsub is configured via the Address Space, so it should be doable by handing the request/responses.

Note that if you instead mean the normal Subscription system via the MonitoredItem/Subscription ServiceSets, that is doable right now, however you still need to handle all the logic yourself. There is also one example that might help https://github.com/OPCFoundation/UA-Java/tree/master/examples/publisher.

sergeyk183, 2018-09-10:
can you share a MonitoredItem example please?

bjakke, 2018-09-13:
Not possible, it is _way_ outside the scope of the functionality of this stack, even for examples. The stack only handles the communication layer, you are expected to know OPC UA well enough to do rest yourself if you are using just the stack level. I recommend using proper SDK level products (search e.g. https://opcfoundation.org/products) instead.

But generally please read specification parts 1,3,4. In part 4 it lists the Subscription and MonitoredItem ServiceSets, you will need to handle the 10+ different service calls, and handle states of the items and have a logic for queue:ing and handling the PublishRequests.

closed 2018-09-13 by bjakke

pradeipk, 2021-08-17:
@bjakke, @hkarabakla   We are using this Java stack 1.4 for our client-server implementation. We were wondering if we can use the same for PubSub support, but based on the comments above seems that the java stack does not support OPC specs 1.04. Please confirm if this is still the case (rather than asking in a new issue I am asking the query in this thread itself) 
cc @jouniaro

jouniaro, 2021-08-17:
Unfortunately, although the stack has nowadays basic support for OPC UA 1.04, it does not include the necessary transport implementations for UDP, MQTT or AMQP that PubSub requires.