OPCFoundation/UA-Java-Legacy #156: Can the SampleClient show more comprehensive Read&Write code?

V4Friend, 2018-08-18:
Dear devs,

I am Industrial Software Engineer with 20+ years of experience writing PLC-code, SCADA and DCS applications etc. The company I work for develops its own JAVA-based MES-system. 

To be "ready for the future" we are looking to implement our own OPC-UA-client, so we can communicate with all brands of PLCs that support OPC-UA (in the near future).

IMO, OPC-UA offers a number of features which make it "better" than classic OPC:

- Reading and writing STRUCT data;
- Reading and writing ARRAYs of data;
- Subscribing to data, so you get fresh data from the PLC without polling;

Now I have a few questions, after examining the UA-Java stack code:

- I assume that the OPCFoundation UA-java stack supports all these features? 
- Are there limits, to the amount of data in the STRUCT or ARRAY reads/writes?
- In the examples, I don't seem to find any code which implements these advanced features. Could that be added to the Client example? It would really increase the usefulness!

Thanks in advance.

jouniaro, 2018-08-20:
The UA-Java stack is just the communication stack, providing an interoperable implementation of OPC UA messaging. The examples are very basic and for anything more advanced, you will need an OPC UA SDK, which can provide a higher level API and therefore also enables higher level examples. 

Subscriptions especially require quite much from the application level and therefore require an SDK level to manage the proper handling.

To answer your questions: yes, the stack supports all of these features. Usage of arrays is pretty straight forward. Structures as well, except that in order to support your own types, you will need to configure the types into the stack's message encoder/decoder. This is also made way easier in the SDK level.

closed 2018-08-20 by jouniaro

V4Friend, 2018-08-20:
Thanks for your answer. Though it raises new questions:

- Which SDK (s) do you recommend?
- If you buy an SDK, you don't need to become member of the OPCFoundation?

jouniaro, 2018-08-21:
[Prosys OPC](https://www.prosysopc.com) has a mature SDK that is built on top of this stack and can help you also with custom structures very well. Prosys OPC (including me) is also the main developer of this stack.

And yes, if you buy an SDK, you will get the right to use the stack in your commercial products as well.

V4Friend, 2018-08-21:
Thanks again! It's a good thing that the license is "per developer". I will pass it on to the one(s) holding the wallet :)