OPCFoundation/UA-Java-Legacy #157: Illegal reflective access warning when running on java 10

bjakke, 2018-08-21:
The following warning can be seen in system error when running with java 10.0.2

> 
> WARNING: An illegal reflective access operation has occurred
> WARNING: Illegal reflective access by org.opcfoundation.ua.utils.ObjectUtils (<PATH to classes>) to field java.lang.Enum.name
> WARNING: Please consider reporting this to the maintainers of org.opcfoundation.ua.utils.ObjectUtils
> WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
> WARNING: All illegal access operations will be denied in a future release

This will probably require some rework in how Structures toString is implemented or some logic change in ObjectUtils helpers that print the entire fields of instances reflectively and recursively. This might be more important depending on what is the default logic in Java 11.

bjakke, 2018-09-26:
Based on https://jdk.java.net/11, the situation seems to be the same in Java 11.

closed 2018-11-30 by bjakke

AdrianChallinorOsiris, 2020-06-13:
Also seen on Java 13