OPCFoundation/UA-Java-Legacy #172: Reading Kepware system Date tag

debinaik, 2018-10-04:
Hi Team,
It's really useful, I am able to handshake with kepware successfully. My next target is to read kepware system date (i.e kepware tag   " _System._Time_Second"). By any chance you have sample code piece for the same ?

debinaik, 2018-10-04:
Address of "_System._Time_Second" in kepware is  ->  i=2258.
How I can use above address in OPC-UA client ?

debinaik, 2018-10-04:
I got above answer .  We need to use [ Identifiers.Server_ServerStatus_CurrentTime] in below code..
ReadValueId[] nodesToRead = { new ReadValueId(Identifiers.Server_ServerStatus_CurrentTime, Attributes.BrowseName, null, null) };

But how to use the below formatted address  in OPC UA client java code ?
ns=2;s=MqttChannel.SSM1._System._Enabled

Karthikeyyan, 2018-12-05:
@debinaik - Did you get an answer for this - "ns=2;s=MqttChannel.SSM1._System._Enabled"?
If so please let me know. Thanks.