OPCFoundation/UA-Java-Legacy #186: Bad_RequestTypeInvalid code=0x80530000 "The security token request type is not valid."

sashio123, 2018-12-09:
Hi,

We are using Kepware to connect to OPC UA server and exactly after 1 hour of establishing the connection we are facing the following error :

o.o.ua.transport.tcp.io.TcpConnection /***.**.***.**:49320 Error
org.opcfoundation.ua.common.ServiceResultException: Bad_RequestTypeInvalid (code=0x80530000, description="The security token request type is not valid.")
                at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run(TcpConnection.java:782)
                
and after every 30 mins of retrying it is throwing the below error :

2018-12-05 10:23:44,995 ERROR [Blocking-Work-Executor-40] o.o.ua.transport.tcp.io.SecureChannelTcp -560875045 Failed to renew security token. 
org.opcfoundation.ua.common.ServiceResultException: Bad_ServerNotConnected (code=0x800D0000, description="The operation could not complete because the client is not connected to the server.")
                at org.opcfoundation.ua.transport.tcp.io.TcpConnection.sendRequest(TcpConnection.java:1083)
                at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.createSecureChannel(SecureChannelTcp.java:673)
                at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.access$500(SecureChannelTcp.java:97)
                at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp$8.run(SecureChannelTcp.java:1214)
                at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
                at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
                at java.lang.Thread.run(Thread.java:748)

We are using the code wrapped around the opcua foundation code and using it to connect from kepware client.Can provide more details/log further.

sashio123, 2019-01-17:
Hi Team,

Have anyone looked at it by any chance ?

Thanks,
Sashi

bhandari-abhishek, 2019-05-09:
Dear Team,
Has anyone addressed the issue reported?

Thanks,
Abhishek

prasenjith-altizon, 2021-11-24:
Hi Team,
Has anyone checked it?

Thanks,
Prasenjit