OPCFoundation/UA-Java-Legacy #208: Writing to plc takes long

friko16, 2020-09-29:
I need your help, we got stuck in our project because of this.
My issue is that writing to plc takes too long time, around 2-3 seconds.
When writing less values, like 2-3, then it takes shorter around 50ms, so it's seems it's linear dependent on the number of values.
We can't move on with our product, please help.
pseudo-code ( giving you idea how I am doing it ):
`
WriteValue[] _writeValues = new WriteValue[20];
// initializing writeValues with primitive opcua values.....
//_writeValues[0] = new WriteValue(
//                    new NodeId(PlcGateway.namespaceIdx = 3, "NodePath"),
//                    Attributes.Value,
//                    null,
//                    new DataValue(new Variant(new UnsignedLong(2))
//            );
initializeWriteValues(writeValues);
SessionChannel sessionChannel; // initialized
sessionChannel.Write(writeValues);
`

jouniaro, 2020-09-29:
Unfortunately, the OPC Foundation doesn't have any support services available, in fact. But in this case, the problem is 100% on the server side - you cannot perform the write any faster from the client, if you need to write all those changes.

friko16, 2020-09-29:
@jouniaro thanks for reply. The team writing software for PLC is pointing fault to us, the software team whole is the client of PLC. How can we prove to them, that it's an issue on the server side ? Is there a way to check out some PLC settings, e.g. with write speed or something ? Can they provide some PLC hardware info to point out, that it's not possible to write faster ?

jouniaro, 2020-09-29:
You can use Wireshark to analyse the traffic and see how long it takes for the server to respond to your request. See https://www.prosysopc.com/blog/opc-ua-wireshark/ for details, for example.

friko16, 2020-09-29:
@jouniaro but I have measured it already inside my java app, by reading time right before and after writing the values. The PLC team still points to the java lib being the problem. Is measuring time by opc-ua-wireshark any better than my current approach with measuring time with java ?

jouniaro, 2020-09-29:
Well, the Wireshark leaves your client side out of the question completely, so you can point out exactly how it takes for the server to process the request.

friko16, 2020-09-29:
@jouniaro ohh thanks so much for pointing that out. I thought that maybe the problem is that I am sending those values in batch as list using the session.Write() method.
I thought if I write the whole structure as one request ( not one by one in list, with primitive values as in the example in topic), by implementing serializer and deserializer, it would help. So you're saying that's not the case ?

jouniaro, 2020-09-29:
Yes, I understood that you are making one Write call with several WriteValues in it. Then there is not much to optimise on your side.

kevinherron, 2020-10-16:
@friko16 what PLC or server are you connecting to?