OPCFoundation/UA-Java-Legacy #211: Change SSL Certificate thumbprint algorithm to SHA256

pradeipk, 2021-04-01:
We wanted to change SSL certificate thumbprint algorithm from SHA1 to SHA256. For this we made changes in createThumbprint method of CertificateUtils class. and tried to test it with UA CPP and Softing simuation servers.

For UACpp server, we got following exception:
*Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_EncodingLimitsExceeded (0x80080000) "The message encoding/decoding limits imposed by the stack have been exceeded."
at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run*

For Softing, the exception is:
*Exception in thread "main" org.opcfoundation.ua.common.ServiceResultException: Bad_SecurityChecksFailed (code=0x80130000, description="Could not verify security on OpenSecureChannel request.")
at org.opcfoundation.ua.transport.tcp.io.TcpConnection$ReadThread.run*

Looks like we will need further changes to fix this.
@jouniaro  was this tried before? We are not familiar with the java stack code to that extent, can you suggest what else we will need to change in order to fix this. This is reported as a security vulnerability in one of our security scans.

pradeipk, 2021-05-20:
@jouniaro Do you have any idea about this?

jouniaro, 2021-05-20:
As far as I know the thumbprint is not used for security checks. It's just an identifier for the certificate. So you must have some other issue.

pradeipk, 2021-05-20:
Yes. But with SHA1 there are no errors and as soon as we change it to SHA256, we get errors. Is there any way to change the thumbprint algorithm from SHA1 to SHA256?

pradeipk, 2021-05-20:
We think that with SHA256 the thumbprint length increases and when it's being transferred over using TCP it gives the error 'message encoding/decoding limits have been exceeded'. We just changed the createThumbprint method in CertificateUtils.class to use SHA256.
We don't get any error with SHA1.