OPCFoundation/UA-Java-Legacy #221: log4j RCE Exploitation

4N1S, 2021-12-12:
Hello; 
It seems to me that you should investigate on log4j that you use in parallel with crypto-api; There are two important CVE which combined could create a lot of incident on the OPC protocol;

https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2020-0601
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2021-44228

Sincerely

jouniaro, 2021-12-13:
Yes, we have been investigating CVE-2021-44228.

The OPC UA Java Stack itself is not directly vulnerable, since it is using the SLF4J for logging. So the issue depends on how the applications (such as the sample applications) then direct the SLF4J logging.

http://slf4j.org/log4shell.html

The sample applications in the stack are using log4J version 1, so they should not be affected by this issue. However, it is probably affected by other security issues:

https://www.lunasec.io/docs/blog/log4j-zero-day/

The stack is still supporting Java 6, so the samples cannot just update to the latest log4j 2.15.0, although, overall that might be the best. So, I think we will just remove the dependency to log4j altogether and leave it for the users to choose the proper library for their applications.

Issue CVE-2020-0601 does not affect the OPC UA Java Stack, since there is no ECC cryptography implemented in the OPC UA Java Stack, at all. But, thanks for the note for that, too.

jouniaro, 2021-12-17:
I changed the examples to use logback, instead of log4j. Logback is still maintained so it's a safer choice for future.

closed 2021-12-17 by jouniaro

4N1S, 2021-12-17:
sorry but there is also a RCE on logback ;

jouniaro, 2021-12-17:
Do you mean https://jira.qos.ch/browse/LOGBACK-1591, which was fixed in version 1.2.9 - which I used here.

jouniaro, 2021-12-17:
More about that in http://logback.qos.ch/news.html