OPCFoundation/UA-Java-Legacy #236: Build fails when API from higher version Java (7 and above) is introduced in the source code.

pradeipk, 2024-05-13:
Animal Sniffer plugin is used to report API-related incompatibilities at build time. This prevents the buid to succeed when the  APIs from higher version (7and above) of Java is introduced in the source code. Is there any particular reason for this constraint other than JRE 6 compatibility?

If the requirement is to use API from higher version java in the sdk source code, will the PR be accepted? If not, should it be fork and maintain separately? Please help us understand the process of updating and maintaining source code under this requirement.

pradeipk, 2024-05-13:
Thanks @jouniaro, I got it that such PR will not be accepted where API from higher version of Java is used. In this situation can the repo be forked and changes persisted separately. Is there a legal requirement to use the forked repo and not contribute changes back to master?

jouniaro, 2024-05-27:
Please check https://opcfoundation.github.io/UA-Java-Legacy/. 

It says: “Note: Dual license applies to this repository only; GPL 2.0 applies to all other repositories”, which means that you can make changes to a fork which you don't have to return but that must be provided as GPL 2.0.