OPCFoundation/UA-Java-Legacy #241: Issues reported by Security Scan App on Java stack

pradeipk, 2024-12-16:
**Below Issues were reported by Security Scan App on the Java Stack**

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/PRADEE~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/PRADEE~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

file | type | cwe_id | cwe_title | cwe_url | description | severity | language | step_name
-- | -- | -- | -- | -- | -- | -- | -- | --
src/main/java/org/opcfoundation/ua/utils/CertificateUtils.java | Weak   Hash Strength | CWE-328 | CWE-328:   Use of Weak Hash | https://cwe.mitre.org/data/definitions/328.html | 'getInstance'   method of 'java.security.MessageDigest' uses a non-recommended hash   algorithm. | Low | Java* | Weak   Hash Strength
src/main/java/org/opcfoundation/ua/transport/security/PrivKey.java | Missing   Cryptographic Step | CWE-325 | CWE-325:   Missing Cryptographic Step | https://cwe.mitre.org/data/definitions/325.html | 'init'   method of 'javax.crypto.Cipher' uses a crypto algorithm in a wrong way. AES   key is not properly generated | Low | Java* | javax.crypto.Cipher.init
src/main/java/org/opcfoundation/ua/transport/security/PrivKey.java | Insecure   Cryptographic Algorithm | CWE-327 | CWE-327:   Use of a Broken or Risky Cryptographic Algorithm | https://cwe.mitre.org/data/definitions/327.html | 'getInstance'   method of 'javax.crypto.Cipher' uses a non-recommended crypto algorithm. | Medium | Java* | javax.crypto.Cipher.getInstance

</body>

</html>

jouniaro, 2024-12-16:
The first one means we use SHA-1. But that's in the OPC UA specifications and te respective policies are deprecated. But, of course, we need to implement those policies still.

The second  one, I don't understand. Do you have more information, why it's wrong?

The third one is related to the algorithm ("AES/CBC/PKCS5Padding") used to store the private key secretly. I guess, it refers to a possible padding oracle, but this is only related to storing the key in a file (data at rest), so there is no way to use the oracle for it. See the following answer, for example:

https://crypto.stackexchange.com/questions/72444/is-the-java-function-aes-cbc-pkcs5padding-vulnerable-to-padding-oracle

pradeipk, 2025-01-23:
Hi @jouniaro , Further details on the issue reported is as below

- 'init' method of 'javax.crypto.Cipher' uses a crypto algorithm in a wrong way. AES key is not properly generated

**src/main/java/org/opcfoundation/ua/transport/security/PrivKey.java**
_line 210 : cipher.init(Cipher.DECRYPT_MODE, keySpec, ivSpec);_

- 'update' method of 'org.bouncycastle.crypto.signers.PSSSigner' uses a crypto algorithm in a wrong way. Signer updation w/o valid initialization.

**src/main/java/org/opcfoundation/ua/transport/security/BcCryptoProvider.java**
_line 268 signer.update(dataToSign, 0, dataToSign.length);
line 310 : signer.update(dataToVerify, 0, dataToVerify.length);_

To my assumption the scan tool is talking about signer.init(forSigning, params)  which is invoked before the update method in `getAsymmetricSigner` method at line 407.

Kindly share your feedback on these matters—do you consider them noise or genuine issues that need to be addressed? Additionally, we have raised PRs for some of the issues identified by the scan tool. Please review and approve the PRs.

pradeipk, 2025-01-23:
https://github.com/OPCFoundation/UA-Java-Legacy/pull/244#issuecomment-2609459266